Louisvdw/dbus-serialbattery #380: file "revov.py" missing after installation

Markusenz, 2022-12-30:
When installing with the commands
wget https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/installrelease.sh
sh installrelease.sh

the file "revov.py" was missing, I did that on three raspberry's and all showed this problem.
after copying that file manually to the directory /data/etc/dbus-serialbattery/ and starting install script again, all worked ok.
Seems the file is missing in the current install release?

Louisvdw, 2022-12-30:
Hi
There was a new release published yesterday that should fix that issue. Are you using V0.14.3 or something older?

Markusenz, 2022-12-30:
yes it was 0.14.3

mr-manuel, 2023-03-09:
@Markusenz the issue should be solved. Can you please close the issue if so?