Louisvdw/dbus-serialbattery #396: Custom device name lost after reboot

gurkc006, 2023-01-07:
**Describe the bug**
after entering a custom device name (e.g. "Battery") in the GUI under /device/name and a reboot, the default device name (e.g. "SerialBattery(JKBMS)") is set again

**To Reproduce**
Steps to reproduce the behavior:
1. set device name in GUI in Device->Name->"Battery"
2. reboot Venus OS
3. see error

**Expected behavior**
A user defiend device name should persist a reboot of the Venus OS

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi, Venus OS 
 - Venus OS Version v3.00~14

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JKBMS
 - Cells: 16
 - Interface: USB-RS232

Louisvdw, 2023-01-09:
It's a known issue, thanks. Will most likely be fixed when there is a GUI for the driver setup in a next release.

gurkc006, 2023-01-10:
Hi Loiusvdw, 
thanks for the response and good to hear, that it's already on the way to be solved.

mr-manuel, 2023-03-09:
This can be closed.

Reference to existing issue https://github.com/Louisvdw/dbus-serialbattery/issues/100

closed 2023-04-14 by mr-manuel