Louisvdw/dbus-serialbattery #397: Wrong cell voltage reading Daly BMS

ats-rozruch, 2023-01-07:
**Describe the bug**
I have 12S Daly BMS and function read_cells_volts always returning from here

if cells_volts_data is False:
logger.warning("read_cells_volts")
return False

I have to change 
cellvolt_buffer = b"\xA5\x40\x94\x08\x00\x00\x00\x00\x00\x00\x00\x00\x82\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
to
cellvolt_buffer = b"\xA5\x40\x94\x08\x00\x00\x00\x00\x00\x00\x00\x00\x82"

becuse BMS doesn't want to answer to that huge command

And second thing

towatai, 2023-02-05:
Thanks, the "cellvolt_buffer" fixed the problem for me with no cell voltages being read out on my Daly 24V, 8S via RS485. Via UART it worked without problems.

transistorgit, 2023-03-06:
Thanks, I had the same problem. I have 2 Dalys, one by RS485 and one by UART. The RS485 one was not working. 

- The old size of cellvolt_buffer was meaningless, as it is a buffer for the command, that is 13 bytes always. 
- The maxframe count was no problem for me as I have 16s, but I confirm that your fix is right
- lenFixed was also a bug, confirmed

With all these fixes, my daly with RS485 is working fine now. I use it together with [AggregateBatteries](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries) and that would not work without the cell voltages.

As an explanation why the big buffer is a problem for RS485 but not for UART, I think serialbattery still sends the buffer bytes when Daly already starts to reply. With UART this is ok, because its bidirectional. But RS485 is half-duplex so the datastreams collide and the reply is unreadable.

Further, there is still another problem. After reading cell voltages, the next operation is read_soc_data and the first try always fails. So this leads to a log file flooded with warnings "ERROR:SerialBattery:>>> ERROR: No reply - returning".
This happens with both interface types. 
I tried a quick fix with sleep(0.5) between calls but no success. As the read function always flushes the buffers, I don't know where the influencing is. The commands on the serial port for read_soc_data are always the same (a540900800000000000000007d), but on the first try after read_cells_volts the Dalys just don't answer anything. 
Can somebody confirm this behavior? @towatai @ats-rozruch ?

transistorgit, 2023-03-07:
@ats-rozruch do you plan to make a pull request? if not i would do

transistorgit, 2023-03-08:
seems that I was able to also fix the "No reply- Returning" bug, and improved some other bugs. I will create a pull request together with your suggestions if you don't mind.

mr-manuel, 2023-04-20:
@transistorgit  is this issue fixed with your PR or is still something open?

transistorgit, 2023-04-20:
yes, I think it's fixed. Can only test with 2 16s systems, though. Runs fine for a month now.

PR #484 (2023-05-02): Fix #397 wrong cell voltage reading daly bms
this pull requests addresses some bugs found inside daly.py and utils.py, and were inspired by the findings of @ats-rozruch

- the command buffer was much to big and caused problems with RS485 communication
- the buffer length calculation was not working for all cell string legths
- some other wrong numbers in check sum length, field length etc
- checksum check was missing for cell voltage read function
- in utils.py was a wrong calculation of packet lenth that didn't took the different behaviour of usages with and withoud fixed package length

fixe [... cut]
commit da4bc764 (2023-05-02): Fix #397 wrong cell voltage reading daly bms (#484)

* fix daly rs485 communication for cell voltages

* fix read_serialport_data length calculation, fix daly cell voltage packet offset calculation, add checksum check also. Harden general package validity check.

* revert read_serialport_data() to the original state

* copy read_serialport_data() to daly.py

* fix missing bracket

* black reformatting

---------
closed 2023-04-20 by mr-manuel