Louisvdw/dbus-serialbattery #400: EG4/Lifepower 12v/400ah battery not communicating

Justice12t, 2023-01-08:
While viewing the program scan for the BMS, it gets to Lifepower and returns "incorrect data" It doesn't appear any services start as none of the devices (MTTPs, ect) go into ext control.

1. Login to root
2. run tail -f /data/log/dbus-serialbattery.ttyUSB0/current
3. As it scrolls through the list of BMS' it gets to lifepower
4. "incorrect data"
 
**Expected behavior**
Battery/BMS information showing within the Cerbo GX, connected devices (MPPTs, ect) go into ext control.

**Screenshots**
I can grab one if needed

**VenusOS (please complete the following information):**
 - Cerbo
 - 3.00 ~14

**Battery/BMS (please complete the following information):**
 - Tian Power (LifePower/Revov) assuming it's the same BMS (with instructions I could pull the battery apart and look)
 - Cells: 4
 - Interface: USB-RS485

**Additional context**
No errors during install.  Using the same cord that was supplied with the batteries that allows me to connect the battery to a PC and use the provided (by signature solar) BMS software to view the stats.  Currently the batteries are set via dips switches to #1 and #2.

Justice12t, 2023-01-08:
Hopefully, I'm assuming correctly that my battery should work seeing the list states Tian Power (LifePower/Revov).  If I'm wrong please let me know.

Justice12t, 2023-01-08:
Just reading through old posts.  I've got the 2 batteries connected together and then 1 cable to the Cerbo...any chance this could be the reason?

Louisvdw, 2023-01-09:
This is your problem. Each BMS has to be connected using it's own cable.
I have not yet tested with firmware 3.00~ so that might also be an problem but I don't know.

Rather use firmware 2.92 (latest release) and connect one battery and see if that works.

Justice12t, 2023-01-09:
I figured it was user error.  My apologizes.  I've already purchased another cable for the second battery.  I'll down grade to the latest release 2.92 and test again tonight.  Thanks for the help.

Justice12t, 2023-01-09:
Ok back to 2.92...single battery...fresh driver install...same issue...what else am I missing. 

[image]
[image]
[image]

Louisvdw, 2023-01-10:
The driver is installed and is running through it's tests. So that part is all working fine. 
It does not find a battery. This is in most cases a problem with the wiring or with the adapter used. So either a loose connection, or the BMS is using a different UART than the adapter you are using. 
So check that these match. There are 2 common UART types used. TTL or RS485 (sometimes also RS232)

Justice12t, 2023-01-10:
Am I safe to assume that if this cable works connecting the battery (rs485) to a PC (usb) reading the BMS information using the provided BMS program that the cable should be good?  I've ordered a second that should be here in a few days just to double check.  I'm going to try a few other tests tonight and see if I can get it working or narrow down the challenge.

Louisvdw, 2023-01-10:
Yes. If the cable works to a PC it should work for the driver to the GX as well

Justice12t, 2023-01-11:
Tested the cable on both batteries with the BMS tool and both communicated without issue.  Used this cable to connect 1 battery to the Cerbo, removed all other USBs and made sure the batteries were isolated from each other.  Pushed OS v2.92 again, reinstalled the serial program.  No change.  Same output as above.  I tired both batteries and both RS485 ports on each, no change.  Could this battery(EG4 LifePower) be different than the others? Meaning possibly a different BMS than the 24v 200ah or the 48v 100ah??  I'm not sure of a good next step, anything else you'd like me to test?

pchiquit, 2023-01-11:
I have the EG4 24v and the BMS information is displayed on the PC program. Mine says: 3QT-YS00-16SV100A-V3.5. Do you have something similar? That will give you an idea if the BMS is the same.

[image]

Justice12t, 2023-01-11:
Looks like mine is D-4S400AH-P00-ZTR-V1.0 and BMS Version Z4SR02.  That's me reading it from a log file I had a screen shot of.  I'll connect to the batteries tomorrow and confirm this information.

pchiquit, 2023-01-11:
@Justice12t I have documented the EG4 24V protocol on this closed issue #212  . As you can see the EG4 24V uses the preamble `7e 01 01 00 fe 0d` compared to yours that seems to send `01 03 00 00 00 ..` , so the BMS seems to be different. I don't know how technical you are but the process of adding support for your battery is not that complex but would need some debugging from your part to see how close the protocol is from the EG4 24V.  This can be done by playing with the PC program and checking what commands are sent and received and comparing them to what is documented on my old issue.

pchiquit, 2023-01-11:
@Justice12t I have documented the EG4 24V protocol on this closed issue #212  . As you can see the EG4 24V uses the preamble `7e 01 01 00 fe 0d` compared to yours that seems to send `01 03 00 00 00 ..` , so the BMS seems to be different. I don't know how technical you are but the process of adding support for your battery is not that complex but would need some debugging from your part to see how close the protocol is from the EG4 24V.  This can be done by playing with the PC program and checking what commands are sent and received and comparing them to what is documented.

Justice12t, 2023-01-11:
That's what I was afraid of.  But ok fair.  I'm fairly technical and can usually trip through stuff.  I'll check out #212 and see where/if I can go from there.  Thanks for the help on this.

Justice12t, 2023-01-12:
Would it be better if I just opened up a request to have this BMS added and the "battery" tag can be added to it?  I've found the attached document that looks like it breaks down the communications, I'm not sure if it is helpful to the right folks or not.  I did spend some time looking and know some of the responses.  I can add that information to the request if needed.  Let me know, thanks.
[work_Communication.txt](https://github.com/Louisvdw/dbus-serialbattery/files/10405979/work_Communication.txt)

Justice12t, 2023-02-14:
@pchiquit @Louisvdw 
I attached my break down of my BMS for the EG4 12v/400.  Hopefully it makes sense.  I don't have all the answers but figured folks much smarter than me might be able to lend a hand.  I still have a few fields at the end to figure out which hopefully I'll get to this week.  I used the screenshot I posted above for the information.  Let me know if this is helpful or a complete waste of time.  Thanks!
 
[BMS Breakdown.txt](https://github.com/Louisvdw/dbus-serialbattery/files/10728387/BMS.Breakdown.txt)
[EG4-LL-MODBUS-Communication-Protocol_ENG-correct-1.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/10728389/EG4-LL-MODBUS-Communication-Protocol_ENG-correct-1.pdf)

pchiquit, 2023-02-16:
@Justice12t let me look and see if I can add this. I'll need a few days to find time to work on that.

Justice12t, 2023-02-17:
Awesome!  Thank you.  I've got a few minor updates.  The full capacity looks like it needs to be converted 1440000000 milliampere = 1440000 ampere, 1440000/3600 = 400 AH.  Also it starts with 01 03 4E... the 01 seems to correspond with the dip switch setting, so in this case it's set to 1 or battery 1.  Sure you already know this but just wanted to make sure I updated the info.  I'll see if I can work on the model, FW and Serial # this weekend.  Thanks again.  Let me know what else I can do to help.

Justice12t, 2023-02-19:
I updated this file (at the bottom) with the serial #, FW version and Model.  I'm also including a file of the most recent communication detail that I used to figure this out.

[BMS Breakdown.txt](https://github.com/Louisvdw/dbus-serialbattery/files/10778233/BMS.Breakdown.txt)
[work_Communication.txt](https://github.com/Louisvdw/dbus-serialbattery/files/10778234/work_Communication.txt)

Justice12t, 2023-03-12:
@pchiquit just wanted to check in and see how things were going and if there's anything else you needed from me.  Let me know, thanks!

Justice12t, 2023-03-13:
@pchiquit no need to apologize. IF you aren't able to find the time no big deal either as I think I found another route so just let me know.  Thanks!

Deemg003, 2023-03-21:
@pchiquit If you need another tester, let me know. 
Running 3x 12v400ah on Cerbo

Justice12t, 2023-04-16:
@Louisvdw can this get "tagged" with battery?  I'm assuming that's how it gets added to the list to be reviewed and possibly code added? I believe we have all the pieces needed and I just didn't want it to fall off the list.  Thanks!