Louisvdw/dbus-serialbattery #403: Add option - use Victron counter instead of BMS

mr-manuel, 2023-01-11:
**Is your feature request related to a problem? Please describe.**
Unfortunately not every BMS shows the right amount of current drawn, even if calibrated. After calibration either high current values are shown right or low current values. This causes that the Ah counter counts wrong and the SoC is also completely wrong.

**Describe the solution you'd like**
It would be nice if we can choose in the setting file, if the BMS counters (power, current, voltage, Ah) are used or the Victron once.

**Describe alternatives you've considered**
I tried to calibrate my JK BMS multiple times with a professional meter, but no chance that the BMS shows the correct values in low, mid and high current. At the moment I calibrated the BMS at the current which is displayed most of the time.

**Additional context**
https://github.com/Louisvdw/dbus-serialbattery/issues/8#issuecomment-1373966490

LarsAndersenDenmark, 2023-03-28:
For me it is not enough that I shall monitor the BMS values. It should be possible to choose which informations comes from where. I.e. SOC and current comes from the shunt, but cells values comes from the BMS....and all values are used as for active controlling of the Victron system by the GX/serialbattery monitor :-)

Today you can just choose the shunt as active batterymonitor and the values for cells are still shown in the VRM...but not used actively by GX. See for example the Youtube from Offgrid Garage.

pos-ei-don, 2023-03-28:
There is a generic mqtt driver for venusos. With this you can write your own NodeRed Script to choose which data you like to use for every case.

mr-manuel, 2023-03-28:
This can be achieve by using a virtual battery and feed the data over MQTT (from Node-RED and Venus-OS Large for example). It allows you to combine the data you want.

You can find the driver here [venus-os_dbus-mqtt-battery](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery).

closed 2023-03-28 by mr-manuel

LarsAndersenDenmark, 2023-04-15:
> This can be achieve by using a virtual battery and feed the data over MQTT (from Node-RED and Venus-OS Large for example). It allows you to combine the data you want.
> 
> 
> 
> You can find the driver here [venus-os_dbus-mqtt-battery](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery).

1) Does all this also work with feedin excess, where the Victron system do not follow the CCL and therefore mppts just go at full throttle?
2)Any youtube for dummies for this? :-)
BR, Lars

mr-manuel, 2023-04-15:
1. It should. Maybe also this helps you https://github.com/Louisvdw/dbus-serialbattery/discussions/490#discussioncomment-5622965
2. No