Louisvdw/dbus-serialbattery #406: Publish Time To Go to VRM dbus path

Louisvdw, 2023-01-11:
Calculate a Time To Go for the battery capacity and publish to dbus com.victronenergy.battery/TimeToGo

Time to in seconds until battery SOC reaches "SOC relay/discharge floor" value.  Capped at 864000 (10 days) when battery is not discharging.
Discharge floor should be SOC_LOW_WARNING (default value 20)

Louisvdw, 2023-01-12:
Hi @corneel27 
This should be easy ticket to implement if you want to look at it. 
We already have calcs for the TimeToSOC and this would be an inverse of when the discharge will deplete the battery and can be implemented in a similar way (except just 1 value to aim for)

corneel27, 2023-01-13:
Hi Louis,
I investigated the code.
In battery.py from linenr 470 and furtheron:

```    
    def get_timetosoc(self, socnum, crntPrctPerSec) -> str:
        if self.current > 0:
            diffSoc = socnum - self.soc
        else:
            diffSoc = self.soc - socnum

        ttgStr = None
        if self.soc != socnum and (diffSoc > 0 or utils.TIME_TO_SOC_INC_FROM is True):
            secondstogo = int(diffSoc / crntPrctPerSec)
``` 
This function checks already if self.current >0 (**charging**) else (**discharging**)
So I think it is already implemented, please correct me if I am wrong.

Louisvdw, 2023-01-14:
That function does do a calculation to custom SOC points yes, but it publish this to it's own path in dbus.
We need to add a spesific calc hare (perhaps seperate from the custom list) that can call this function to calc the deplete time and then publish the result to the path that victron use so that it will populate this in VRM.

So the above function you can give a list of SOC point that it will calculate the time for each point. The result is published for each pount under /TimeToSoC. So if you gave it 20, 50, 100 there will be time values under /TimeToSoC/20/ /TimeToSoC/50/ /TimeToSoC/100/

We need one spesific point when the battery is deplete (SOC_LOW_WARNING value) and then publish that one value it to /TimeToGo/

corneel27, 2023-01-14:
Hi Louis,

I think (and hope) I created the right additions in dbushelper.py.
I didn't test it (don't know how).
I made a pull request with my additions.

Greetings Cees

closed 2023-01-20 by Louisvdw