Louisvdw/dbus-serialbattery #410: Dynamic Discharge Current Control

ppuetsch, 2023-01-13:
I would like to have a CV discharge behaviour - i.e. if I set a mimium Voltage of e.g. 3.0V/cell, then the multi should draw as much current as is possible so that the cell voltage does not get below 3.0V. 

I implemented this feature already in https://github.com/ppuetsch/dbus-serialbattery/tree/dynamic_discharge_current

The basic idea is: you provide two config properties - min cell voltage and internal battery resistance (i think the name is misleading - it should be better internal cell resistance). 

The algorithm estimates the OC voltage of the lowest cell in the current situation by adding current * internal_cell_resistance to the current cell value. From there, it calculates the possible current using again internal cell resistance. 

One can derive internal cell resistance by: 
monitor the battery in a situation where a sudden load appears (in my example: After turning on the Multi battery rises to 70A and Voltage drops by ~1.4V. So the internal Resistance of the whole battery is 20mOhm. As It's a 16S battery, internal resistance of a single cell is 20mOhm / 16 = 1,25 mOhm.

In my observation, this algorithm has a nicer behaviour if coming close to empty, it only requires two parameters to configure, it does not rely on SoC calculations by BMS (which are not useful in my setup). 

In principle, it works very nicely right now. Im still experiencing the "Zero Battery Charge/Dischager Oscillation" as described in https://github.com/Louisvdw/dbus-serialbattery/issues/407

Sample Implementation:
```
    def calcDynamicDischargeCurrent(self):
        estimated_zero_load_voltage = (
            self.get_min_cell_voltage()
            - self.current * utils.INTERNAL_BATTERY_RESISTANCE
        )
        available_voltage_drop = estimated_zero_load_voltage - utils.MIN_CELL_VOLTAGE
        available_current = available_voltage_drop / utils.INTERNAL_BATTERY_RESISTANCE
        result = max(available_current, 0)
        return result

```

What do you think? If you like it, I can clean up everything and integrate it properly. Maybe one the can get rid of the DCCM_CV part of the driver.

Louisvdw, 2023-01-13:
I assume that each cell resistance would be different depending on what cells are used, capacity and age. Is there a generic way to calculate this that we can use in code or would the user have to measure this with a multimeter?

ppuetsch, 2023-01-13:
I did measure/monitor the voltage drop using the VenusOS. 
The proposed algorithm is anayways not 100% accurate - that means it won't nail down the cell voltage to the target cell voltage, which is also caused by the fact that the multi won't set the exact current that the BMS allows. Therefore, probably a default value (e.g. 2mOhm) probably should anyways do the trick, and "expert users" can adapt it if they want to.

Louisvdw, 2023-01-13:
OK that could work.  If you want I can create a beta release that people can use to test it and give feedback?

ppuetsch, 2023-01-13:
Ok - I will make a suggestion in the comming days

mr-manuel, 2023-04-14:
@ppuetsch any update on this? In the meanwhile a lot changed, so I don't know, if this is still relevant.

closed 2023-05-01 by mr-manuel