Louisvdw/dbus-serialbattery #414: Possibility to Contribute to Documentation

ppuetsch, 2023-01-13:
Currently, the documentation lives as Wiki. This is a bit unfortunate, as one can't contribute to that easily. 

One solution might be to copy over the documentation into the repository itself. Then 
a) One can adapt documentation along with code changes
b) has always the documentation that fits the currently checked out revision

What do you think?

Louisvdw, 2023-02-01:
@ppuetsch that is a good idea.
I was planning to move the docs from the wiki to github pages. The markdown does have a few limitations what you can display. 
Do you think normal markdown docs inside the repo/branch will work better than github pages? We could also split it.
I'm open for suggestions, but I do want to expand the documentation with a bit more detail for each BMS type

mr-manuel, 2023-04-14:
I have prepared a solution for this with Docusaurus, GitHub Actions and GitHub Pages. The source code of the docs will be in a folder of the repository and therefore also browsable by version.

mr-manuel, 2023-04-20:
The documentation is now located in the source code in the folder `docs/docs` and is published automatically after new commits to https://louisvdw.github.io/dbus-serialbattery/

closed 2023-04-20 by mr-manuel