Louisvdw/dbus-serialbattery #418: Wrong device install

Marccatamaran, 2023-01-15:
Hi
On the startup of my cerbo gx  v2.92 with these devices connected on the USB 4 ports.
Port 1 daly bms
Port 2 daly bms
Port 3 empty
Port 4 usb gps
 I get a serialbattery (ant ) from Port 4 install ?
[image]

But after the boot is completed of the cerbox gx , I  disconnect the usb gps and re-connect it. This results in the gps device back on. Then I can remove the disconnected device (serialbattery)from the list.

But everytime I do a reboot of the cerbo Gx will the 3 ports  usb connected , the serialbattery (ant) come back again and I lose the gps device.

Regards
[image]

Louisvdw, 2023-01-16:
Hi @Marccatamaran 
What happens is that the GPS response to the same signal as what the ANT BMS use. That is why it is picked up. 

To bypass this as you have to disable the ANT driver. In the current release go to utils.py and comment out the ANT driver in the list of battery types (you will see others like the Sinowealth is already commented out with # in front)
We're busy changing this a bit for future releases so just any new process for the battery types on those.

Once the changes is done, run the following script and restart to apply it.
> sh /data/etc/dbus-serialbattery/reinstalllocal.sh

Marccatamaran, 2023-01-18:
Great thanks

closed 2023-01-18 by Marccatamaran

mr-manuel, 2023-06-11:
@Marccatamaran could you please reproduce this issue and post screenshots of the data that is displayed? Like cell voltages and capacity.

If you are able to a screenshot of the `dbus-spy` values would be great, if not the above screenshots would also help me.