Louisvdw/dbus-serialbattery #421: Wrong Charge Voltage Limit (FLOAT instead of MAX)

MisterX1000, 2023-01-16:
see #415 

After installation of v.0.14.3 FLOAT_CELL_VOLTAGE is used for CVL instead of MAX_CELL_VOLTAGE . 

MIN_CELL_VOLTAGE = 2.9
MAX_CELL_VOLTAGE = 3.45
FLOAT_CELL_VOLTAGE = 3.35

With v.0.14.2 the CVL limit works like expected!

[image]

Here you can see that I installed the v.0.14.3 at 12.Jan - there was the CVL "jump" down.

[image]

morres83, 2023-02-28:
Are there any updates on this topic? In my understanding, when looking at the code, linear mode is not usable as it will only charge up to the FLOAT voltage instead of the MAX voltage.

pkali1, 2023-03-01:
I think there is some kind of time test missing in section manage_charge_voltage_linear of Battery.py.
It should be like in manage_charge_voltage_step

tDiff = time() - self.max_voltage_start_time
if utils.MAX_VOLTAGE_TIME_SEC < tDiff:
self.max_voltage_start_time = None
self.allow_max_voltage = False

Please could someone control, if I am right?

pkali1, 2023-03-02:
Inside battery.py change   FLOAT_CELL_VOLTAGE to MAX_CELL_VOLTAGE  and linear might be working.
I will test it today, if there is a sun..

 def manage_charge_voltage_linear(self):
        foundHighCellVoltage = False
        if CVCM_ENABLE:
            currentBatteryVoltage = 0
            penaltySum = 0
            for i in range(self.cell_count):
                cv = self.get_cell_voltage(i)
                if cv:
                    currentBatteryVoltage += cv

                    if cv >= PENALTY_AT_CELL_VOLTAGE[0]:
                        foundHighCellVoltage = True
                        penaltySum += calcLinearRelationship(cv, PENALTY_AT_CELL_VOLTAGE, PENALTY_BATTERY_VOLTAGE)
            self.voltage = currentBatteryVoltage    # for testing
        
        if foundHighCellVoltage:
            # Keep penalty above min battery voltage
            self.control_voltage = max(currentBatteryVoltage - penaltySum, MIN_CELL_VOLTAGE * self.cell_count)
        else:
            self.control_voltage = MAX_CELL_VOLTAGE * self.cell_count      #  ............may be this is the problem........

morres83, 2023-03-02:
I think this will not fully sove the issue. By doing this, you would most probably be able to charge to MAX voltage, but it will never switch to FLOAT, which is also not the desired behavior. 
I guess the code from `manage_charge_voltage_step(self)` which seems to be working for me needs to be used as basis for MAX-FLOAT switchover.

MisterX1000, 2023-03-02:
> Do we really need floating after some max_voltage time?

Exactly what I thought - If we use LiFePo4 cells it is unnecessary ;-)

morres83, 2023-03-02:
The step mode does reduce to the float value currently and certainly it is out of the LiFePo4's comfort zone if you keep the high voltage forever.
I have to say that I don't understand currently how exactly this PENALTY thing is supposed to be working, if this is ensuring that the high voltage is reduced after some time, it could be okay. But if the cells are not perfectly matched I think high voltage should be kept for some time and the reduction should not occur immediately after one cell hits the high voltage (i.e. to also give the balancer some more time). But as I said, maybe the PENALTY logic is doing all this correctly and the only thing to change then is what pkali1 suggested... ?!

pkali1, 2023-03-03:
Bug repair tested and working. (without switching to float) Makes my home solar every day charging cycle more smooth and fluent. Take care with setting penalties. If someone adds lineard tDiff for floating floating, it would be perfect. 

Here is my example of setting (16x Lifepo4 200Ah) .

# Set Steps to reduce battery current. The current will be changed linear between those steps
CELL_VOLTAGES_WHILE_CHARGING         = [3.55, 3.50, 3.45, 3.30]
MAX_CHARGE_CURRENT_CV                = [   0,    10,  30,  50]

# if the cell voltage reaches 3.55V, then reduce current battery-voltage by 0.01V
# if the cell voltage goes over 3.6V, then the maximum penalty will not be exceeded
# there will be a sum of all penalties for each cell, which exceeds the limits
PENALTY_AT_CELL_VOLTAGE  = [3.47, 3.55, 3.6]
PENALTY_BATTERY_VOLTAGE  = [0.01, 0.05, 0.1]  # this voltage will be subtracted

Thanks and I think you can pull this simple bug repair into code.

pkali1, 2023-03-04:
Floating added. For testing.

`    def manage_charge_voltage_linear(self):
        foundHighCellVoltage = False
        if CVCM_ENABLE:
            currentBatteryVoltage = 0
            penaltySum = 0
            for i in range(self.cell_count):
                cv = self.get_cell_voltage(i)
                if cv:
                    currentBatteryVoltage += cv

                    if cv >= PENALTY_AT_CELL_VOLTAGE[0]:
                        foundHighCellVoltage = True
                        penaltySum += calcLinearRelationship(cv, PENALTY_AT_CELL_VOLTAGE, PENALTY_BATTERY_VOLTAGE)
            self.voltage = currentBatteryVoltage    # for testing

            if None == self.max_voltage_start_time:     # zacatek noveho kodu
                if MAX_CELL_VOLTAGE * self.cell_count <= currentBatteryVoltage and True == self.allow_max_voltage:
                    self.max_voltage_start_time = time()
                else:
                    if SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT > self.soc and not self.allow_max_voltage:
                        self.allow_max_voltage = True
            else:
                tDiff = time() - self.max_voltage_start_time
                if MAX_VOLTAGE_TIME_SEC < tDiff:
                    self.max_voltage_start_time = None
                    self.allow_max_voltage = False      # konec noveho kodu`

pkali1, 2023-03-04:
def manage_charge_voltage_linear(self):
        foundHighCellVoltage = False
        if CVCM_ENABLE:
            currentBatteryVoltage = 0
            penaltySum = 0
            for i in range(self.cell_count):
                cv = self.get_cell_voltage(i)
                if cv:
                    currentBatteryVoltage += cv

                    if cv >= PENALTY_AT_CELL_VOLTAGE[0]:
                        foundHighCellVoltage = True
                        penaltySum += calcLinearRelationship(cv, PENALTY_AT_CELL_VOLTAGE, PENALTY_BATTERY_VOLTAGE)
            self.voltage = currentBatteryVoltage    # for testing

            if None == self.max_voltage_start_time:     # zacatek noveho kodu
                if MAX_CELL_VOLTAGE * self.cell_count <= currentBatteryVoltage and True == self.allow_max_voltage:
                    self.max_voltage_start_time = time()
                else:
                    if SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT > self.soc and not self.allow_max_voltage:
                        self.allow_max_voltage = True
            else:
                tDiff = time() - self.max_voltage_start_time
                if MAX_VOLTAGE_TIME_SEC < tDiff:
                    self.max_voltage_start_time = None
                    self.allow_max_voltage = False      # konec noveho kodu

pkali1, 2023-03-04:
Floating time added. Not complete testing.

pkali1, 2023-03-06:
The code from battery.py was not complletely copied.. sorry. Now corrected and working correctly with driver 14.3 and venus 2.90.
Regards

mr-manuel, 2023-03-12:
Someone knows, why this was changed?

https://github.com/Louisvdw/dbus-serialbattery/commit/3685d15a5d1ba801867bbdb98f57fd805ea2c24a#diff-9d9b25b4a5dcc9b6f162c96264601c1986789dd796a6b170586076a746fabbc0R126-R130

mr-manuel, 2023-03-12:
> The code from battery.py was not complletely copied.. sorry. Now corrected and working correctly with driver 14.3 and venus 2.90. Regards

@pkali1 you are right, that the time counter for the max voltage is missing.

I added this to my fork. Please try it and let me know. https://github.com/mr-manuel/venus-os_dbus-serialbattery/tree/issue-421

You can install it by executing this commands:

```bash
cd /tmp

wget https://github.com/mr-manuel/venus-os_dbus-serialbattery/archive/refs/heads/issue-421.zip
unzip issue-421.zip

cp -rf /tmp/venus-os_dbus-serialbattery-issue-421/etc/dbus-serialbattery/ /data/etc

chmod +x /data/etc/dbus-serialbattery/*.sh
chmod +x /data/etc/dbus-serialbattery/*.py
chmod +x /data/etc/dbus-serialbattery/service/run
chmod +x /data/etc/dbus-serialbattery/service/log/run

bash /data/etc/dbus-serialbattery/reinstalllocal.sh
```

Pay attention, that in this release the configuration is not done anymore in the `utils.py`, but in a custom `config.ini`. Just copy the `config.sample.ini`. This prevents overriding the settings after updating in future versions. It includes also a lot more changes.

Bonzai10, 2023-03-13:
hello, 
unfortunately no change. it takes allways the info from step, not linear

[image]
[image]

mr-manuel, 2023-03-13:
Is this enabled?

https://github.com/Louisvdw/dbus-serialbattery/blob/6471152bf701eb46beaa9ba4d11d93ffd1559dd4/etc/dbus-serialbattery/default_config.ini#L88-L89

Currently the config file is a bit confusing, I will fix that.

Bonzai10, 2023-03-13:
no changing :-(

[image]

[image]

Bonzai10, 2023-03-13:
I think the whole linear is not activated

mr-manuel, 2023-03-13:
Have you a voltage curve how the linear and how the step mode should look? I need some more days for further testing. The curve would help me a lot.

Bonzai10, 2023-03-13:
No, after the install of your Version the data is lost 👎

mr-manuel, 2023-03-13:
In VRM Portal? Check if the VRM ID changed

mr-manuel, 2023-03-13:
> no changing :-( i changed MAX_CELL_VOLTAGE = 3.50
> 
> [image]
> 
> [image]

If you changed MAX_CELL_VOLTAGE to 3.50 then this should be right. 3.50 * 16 = 56.00 or am I wrong?

Bonzai10, 2023-03-13:
But i chosed linear, the changing was under step. Therefore i belive that linear is not avaliable. I check tomorrow the VRM again. I believe the usb Port hast changed. This Bug comes with changing from 14.2 to 14.3.

mr-manuel, 2023-03-14:
@WaldemarFech maybe you can help us with that? How should the linear mode you implemented work exactly?

WaldemarFech, 2023-03-14:
Sorry - where's the beginning of the "problem"? 😅

short:
the linear mode was to reduce the charge and discharge current limits - instead of jumping around while reaching the next step.
- but no linear change of any max. charging voltage - this is, as the word say, the maximum... so i would set this to 3.6V (because in my case the BMS resets to 100% when one cell reaches 3.6V). The LiFePo4 cells have absolutely no problem with 3.65V - but you should not go over 3.75V. So there should be everything okay with 3.6V - this will be reached just a minute or less after 3.6V (depending on your charge current of course).

anyway:
the Victron and other charger are not made to measure currents below 2A. So if we calculate a charge or discharge current limit of less than 2A, let's say 0A - then it should stop (we also send the stop charging command).
BUT ... the chargers goes then in the "ok, let's hold the actual voltage then ..."-mode .... and while it tries to hold this voltage (absorption Phase), then there still flows current into the battery - while it shows 0A - wrong!

What were my solution? I've implement the penalty mode.

You can set a penalty voltage. IF one cell goes over a specific voltage, THEN it means, that the actually "absorption" voltage is too high and there is still a flow of current into the battery. So? So we have to reduce the maximum charge voltage until the cell voltage isn't increasing anymore. But how much have we to lower the voltage? Well - until it stops rising. So as more cells goes over the maximum cell voltages and as higher they go over this maximum, as much the penalty will reduce the CVL.

check? :)

I haven't looked into the newer versions anymore - so i don't know what was changed.

can you post your relevant settings in your utils.py and a short description of your bug or the behavior that you would expect?

pkali1, 2023-03-14:
Here are my driver v14.3 and Venus 2.90 with Daly BMS working files. I hope it helps you.

[battery.py.txt](https://github.com/Louisvdw/dbus-serialbattery/files/10964909/battery.py.txt)
[utils.py.txt](https://github.com/Louisvdw/dbus-serialbattery/files/10964910/utils.py.txt)

Bonzai10, 2023-03-14:
Hello,
the error arose when changing from 0.14.2 to 0.14.3. there were moved settings from utils.py to battery.py. Since this version it doesn't work for me anymore. Yours (WaldemarFech 0.14.3) works wonderfully.

however, all the innovations would be "nice to have" as well as the Bluetooth version

WaldemarFech, 2023-03-14:
k, I've did a short overflight from the very beginning xD
@pkali1 said something about the floating voltage - this must be done in any update - i don't know anything about it.

in my opinion the battery should stay as long in the maximum cell voltage as possible - only reducing the voltage if one cell goes over.

why?

the balancer should only work from 3.45V. Everything below is too linear to balance correctly. we want to synchronize the cells at 100%, this begins when the cell voltage is suddenly rising from 3.45V to 3.65V.
But there is nearly no capacity between 3.45 and 3.65V, so the voltage drops fast while balancing.

if we would just go to the max voltage and then lowering the voltage to float voltage, then the balancer would work 10-30mins and every high cell is at 3.45V again. wow. but there are still some cells with 3.4V ... and there is a lot more capacity left and takes hours with a 2A balancer.

So the voltage should kept at 3.6V until all cells reaches e.g. 3.5-3.6V ... then we can definitely say, that they are top balanced. first THEN the system can go in floating voltage. Because now we are finished balancing.

Maybe i will implement this in a fork.

pkali1, 2023-03-14:
Hi. I know LiFePo4 does not need floating, but it is still under discussion and some people say, that it would be better not to hold battery at Max voltage for long time. There is a possibillity in MAX_VOLTAGE_TIME_SEC - so if someone wants to flow for hours, he can set longer time. Anyway thanks for the nice "linear" addon.

Bonzai10, 2023-03-14:
For my part, I could do without the floating. but I suspect the new driver doesn't accept the settings from the linear at all.
Another question: if I set Step Float to 3.55, does the driver work with the adjusted (linear) loading stage? Loading via SOC is not possible with a JKBMS, that's not true at all

WaldemarFech, 2023-03-14:
> Inside battery.py change FLOAT_CELL_VOLTAGE to MAX_CELL_VOLTAGE and linear might be working.

ok - I've read the code and found the problem - as @pkali1 found before - somebody wrote in both functions (manage_charge_voltage_linear and manage_charge_voltage_step)
`self.control_voltage = utils.FLOAT_CELL_VOLTAGE * self.cell_count`
instead of
`self.control_voltage = utils.MAX_CELL_VOLTAGE * self.cell_count`

In the manage_charge_voltage_step function is an implementation of some kind of max-voltage-time (MAX_VOLTAGE_TIME_SEC --> default: 900s). After this time the voltage will drop to the float-voltage.

But this isn't implemented in the manage_charge_voltage_linear function - so this IS a Bug and the code must be changed to MAX_CELL_VOLTAGE (dbus-serialbattery-master\etc\dbus-serialbattery\battery.pi Line 184) or the stuff with the max-voltage time has to be implemented there, too.

- What should i do? Impelent max-voltage time in the linear function? Or to lower the voltage when every cell is over a specific voltage.

i think i have an idea to make everybody happy ... i think i'll make a configureable state machine ... xD ... stay tuned

WaldemarFech, 2023-03-14:
> Another question: if I set Step Float to 3.55, does the driver work with the adjusted (linear) loading stage?

i haven't tested / simulated it - but the code looks so, yes.

mr-manuel, 2023-03-14:
So this should be correct? https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/7562c7432988146a19443691a5ae92707116662f/etc/dbus-serialbattery/battery.py#L156-L239

mr-manuel, 2023-03-15:
I think I understood now correctly the problem and fixed it. I also made the config file much clrearer, since before different options wear spread across the file.

If you could help me with testing would be awesome. It would also be very helpful, if you can track the battery voltage, cell voltages, charge limit voltage and provide a graph of all parameters (see screenshot below).

> I added this to my fork. Please try it and let me know. https://github.com/mr-manuel/venus-os_dbus-serialbattery/tree/issue-421
> 
> You can install it by executing this commands:
> 
> ```shell
> cd /tmp
> 
> wget https://github.com/mr-manuel/venus-os_dbus-serialbattery/archive/refs/heads/issue-421.zip
> unzip issue-421.zip
> 
> cp -rf /tmp/venus-os_dbus-serialbattery-issue-421/etc/dbus-serialbattery/ /data/etc
> 
> chmod +x /data/etc/dbus-serialbattery/*.sh
> chmod +x /data/etc/dbus-serialbattery/*.py
> chmod +x /data/etc/dbus-serialbattery/service/run
> chmod +x /data/etc/dbus-serialbattery/service/log/run
> 
> bash /data/etc/dbus-serialbattery/reinstalllocal.sh
> ```
> 
> Pay attention, that in this release the configuration is not done anymore in the `utils.py`, but in a custom `config.ini`. Just copy the `config.sample.ini`. This prevents overriding the settings after updating in future versions. It includes also a lot more changes.

NOTE: The screenshot is only an example and do not show the data of the working driver.

[image]

Bonzai10, 2023-03-15:
no, takes again the step

[image]
[config.default.ini.txt](https://github.com/Louisvdw/dbus-serialbattery/files/10982737/config.default.ini.txt)

mr-manuel, 2023-03-15:
The linear voltage is only shown/applied, if at least one cell voltage is higher or equal to the first value of `PENALTY_AT_CELL_VOLTAGE = 3.45, 3.55, 3.6` and if `MAX_VOLTAGE_TIME_SEC` is not reached. All other would not make sense in my eyes.

The linear voltage in this case increases/decreases the voltage slowly to not exeed the single max cell voltage and in order that the BMS can balance. If you have no BMS that balances, than this function makes no sense for you.

So, to tell if this is working or not, the screenshot you already provided and a screenshot of the cell voltages is needed when at least one cell voltage is higher than the first value of `PENALTY_AT_CELL_VOLTAGE`.

Bonzai10, 2023-03-15:
[image]

mr-manuel, 2023-03-15:
What are your `PENALTY_AT_CELL_VOLTAGE = 3.45, 3.55, 3.6` voltages?

Bonzai10, 2023-03-15:
[image]

mr-manuel, 2023-03-15:
Have you installed the latest version of my repository? There the `config.sample.ini` is much clearer. It's this part https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/7562c7432988146a19443691a5ae92707116662f/etc/dbus-serialbattery/config.default.ini#L13-L37

mr-manuel, 2023-03-15:
You're right. It did not sync. Now it did :)

Bonzai10, 2023-03-15:
ok. i only need the config.ini, shure

Bonzai10, 2023-03-15:
i belive, no changings :-(

[image]
![grafik](https://user-im
[config.default.ini.txt](https://github.com/Louisvdw/dbus-serialbattery/files/10983272/config.default.ini.txt)
ages.githubusercontent.com/119003116/225401687-a2b9c77f-3037-45ad-8376-4af84593f678.png)

mr-manuel, 2023-03-15:
Can you write me on Discord?

Bonzai10, 2023-03-15:
no, sorry. i don´t know what that is :-(

mr-manuel, 2023-03-15:
Ok, than can you provide me another contact possibility?

Now your cell voltages are under 3.45V. As described above, this won't work, if the voltage is below. Have you checked the `PENALTY_AT_CELL_VOLTAGE`? To be sure which options are set, check with the dbus-spy -> com.victronenergy.battery.ttyUSB0 -> Info/Config/PENALTY_AT_CELL_VOLTAGE. Since there is no more sun you have to load the battery over grid or wait until tomorrow.

Bonzai10, 2023-03-15:
[image]

mr-manuel, 2023-03-15:
You set the first `PENALTY_AT_CELL_VOLTAGE` to `3.48`, that's the problem. If you lower it, it will work.

Bonzai10, 2023-03-15:
but I still haven't reached the penalty at 3.48?

Bonzai10, 2023-03-15:
sorry, i'm not the best supporter :-(.
maybe we should wait until someone with more knowledge (from my side) reports

Bonzai10, 2023-03-15:
i think it takes the values from Charge Voltage limitation and not from Cell Voltage limitation

Bonzai10, 2023-03-15:
sorry, i have to stop now. I'm gone until sunday.
Thank you for your help and effort!!!

Bonzai10, 2023-03-15:
for closing. i go back to waldemarfech
[image]
[image]

mr-manuel, 2023-03-15:
> sorry, i'm not the best supporter :-(.

No problem.

> i think it takes the values from Charge Voltage limitation and not from Cell Voltage limitation

Yes, this is correct. The Cell Voltage limitation is another story. CVL stands for Charge Voltage Limitation and not Cell Voltage Limitation ;)

> sorry, i have to stop now. I'm gone until sunday.

If you are back and the problem persists, let me know how I can contact you, so I can check your system.

Can you send me the whole config you set? So I take the same one for testing.

Bonzai10, 2023-03-15:
at last only changed the config.default.ini like this:
[config.default.ini.txt](https://github.com/Louisvdw/dbus-serialbattery/files/10983700/config.default.ini.txt)

back on waldemarfech it runs how i expect

mr-manuel, 2023-03-15:
For easier troubleshooting I now added "Charge mode" in the GUI

[image]

mr-manuel, 2023-03-16:
> The linear voltage is only shown/applied, if at least one cell voltage is higher or equal to the first value of `PENALTY_AT_CELL_VOLTAGE = 3.45, 3.55, 3.6` and if `MAX_VOLTAGE_TIME_SEC` is not [...]

Here are some more informations: https://github.com/Louisvdw/dbus-serialbattery/issues/297#issuecomment-1328316220

In the latest version of the driver was added, that it goes to float voltage after reached the time threshold.

morres83, 2023-03-16:
Hello Mr. Manuel

I have tested your latest script and I can confirm that it is doing something done on CVL and CCL, but if it is correct I cannot judge. I think it is kind of mix-up between the linear current calculation and votage penalties...
Once the current dropped I realized that my cells were drifting apart very quickly (highest 3.5 , lowest 3.37) and the balancer is trying to do what it can do now (after some weeks of being between 10-40% each day it seems that the imbalance has accumulated very much).

Anyhow , when seeing this, and as I will connect two batteries soon, I think I cannot use the linear mode on both packs and "combine it" somehow in  a virtual battery device. It seems to blurry to me what is under the hood and which values to use in the end. In fact, voltage and current limits are changing second by second....

[image]

mr-manuel, 2023-03-16:
If you use the latest version from my [fork and branch issue-421](https://github.com/mr-manuel/venus-os_dbus-serialbattery/tree/issue-421) you see also more informations about the current state like. It was uploaded just now.

[image]

morres83, 2023-03-16:
OK I will give it again a try with latest version.

Where exactly is it specified  that a discharging by 1 A is done? Is it a Victron "feature"?
In the nonlinear mode normally the CVL is hit when it is reduced to Float. Therefore I understand the discharging. 
But I still don't get it for the penalty, penalty is acting on CVL, isn't it?

mr-manuel, 2023-03-16:
> Where exactly is it specified that a discharging by 1 A is done? Is it a Victron "feature"?

Nowhere. The battery is used as a buffer and when the battery voltage is the same or very close to CVL then this is the normal behaviour of Victron to maintain stability.

> In the nonlinear mode normally the CVL is hit when it is reduced to Float. Therefore I understand the discharging.

No. CVL is reduced in step mode and also in linear mode when the battery voltage is the same as CVL for x seconds.

> But I still don't get it for the penalty, penalty is acting on CVL, isn't it?

Thats true. Please read the linked comment for the penalty. https://github.com/Louisvdw/dbus-serialbattery/issues/297#issuecomment-1327142635

[17 more replies]

commit e762832d (2023-05-06): Some improvements and fixes (#468)

* added balancing switch status

* added balancing switch status

* changed default config file
- added missing descriptions
- changed name

* indentation was wrong

* adding fix for Venus OS >= v3.00~14 QML files

* added cell balancing status for JKBMS

* changed ConsumedAmpHours
should be None instead of 0, if not available

* TimeToSoc fixes
- Added: Show TimeToSoc only, if enabled
- Changed: description in config file
- Changed: default value TIME_TO_SOC_VALUE_TYPE = 1

* resorted functions

* fix CPU overload
- changed: how often Time-To-Soc is calcula [... cut]
commit 25e6fd4c (2023-05-08): Many improvements (#620)

* added balancing switch status

* added balancing switch status

* changed default config file
- added missing descriptions
- changed name

* indentation was wrong

* adding fix for Venus OS >= v3.00~14 QML files

* added cell balancing status for JKBMS

* changed ConsumedAmpHours
should be None instead of 0, if not available

* TimeToSoc fixes
- Added: Show TimeToSoc only, if enabled
- Changed: description in config file
- Changed: default value TIME_TO_SOC_VALUE_TYPE = 1

* resorted functions

* fix CPU overload
- changed: how often Time-To-Soc is calculated in sec [... cut]
closed 2023-04-20 by mr-manuel