Louisvdw/dbus-serialbattery #422: Serial battery driver not found if other VE.direct-USB devices are present

martinpalca, 2023-01-16:
In VenusGX device in bigger installation, there are 6 pcs BMV-700 + 6 pcs MPPT BlueSolar /SmartSolar installed, by use of VE.direct -USB interfaces and active USB 3.0 HUB. 

If JBD BMS with USB-RS485 serial interface is the only device connected to VenusGX, SerialBattery(LLT/JBD) device is found and functional.
If any other USB interface is connected, SerialBattery is not found in any USB port, either native or in USB HUB.

Probably enumerator of USB ports does not account for so much USB connected devices ?

**Screenshots**
[image]

**VenusOS (please complete the following information):**
VenusGX 2.92 (tested both small and large)

**Battery/BMS (please complete the following information):**
JBD BMS AP20S003S , 16 cells , USB-RS485

Louisvdw, 2023-01-16:
Hi @martinpalca 
I assume that you are using the latest release version of serial battery (0.14.3). If not do upgrade to that.
Could you supply me with the serial-starter log file.
https://github.com/Louisvdw/dbus-serialbattery/wiki/Troubleshoot#check-the-log-files-on-your-gx-device

martinpalca, 2023-01-29:
Hi @Louisvdw

it seems that Victron GX device has limit for maximum 8 pcs USB 
connected serial devices. We have removed some USB-VE.direct cable that 
serves for BMV-700, and everything now works.

S pozdravem / Best regards
		Ing. Martin Palča Kolařík, jednatel, CEO, tel. (+420) 608 812 787
		www.ostrovni-elektrarny.cz

Dne 16.01.2023 v 16:01 Louis Van Der Walt napsal(a):

Louisvdw, 2023-02-01:
That is good to know. I'll add the limit to the docs as well.

mr-manuel, 2023-04-15:
I added it to the wiki.

closed 2023-04-15 by mr-manuel