Louisvdw/dbus-serialbattery #425: Device "Name" resetted after system reboot

chiefjoejj, 2023-01-17:
The changeable "Name" of the bms-instance is resetted after a system reboot. Would be great if this could be fixed somehow...

pos-ei-don, 2023-03-08:
As a workaround, I set the Name in the incl des NodeRed at startup (over mqtt)

teefixx, 2023-03-09:
Hi Pos-ei-don
That's in general a good idea, but be careful if the VRM instances change...
see https://github.com/Louisvdw/dbus-serialbattery/discussions/481

pos-ei-don, 2023-03-09:
Why? Mai "Main" battery is a generic mqtt device, filled by NodeRed. So no Problem with this.

teefixx, 2023-03-09:
for the generic device it's no problem. I assume, you always assign the same VRM instance to the generic battery. I thougt, you name your real batteries as well, and there the VRM instance can change...

pos-ei-don, 2023-03-09:
Yes, but NodeRed can "detect" the New id and assign the Name I prefer. So the id doesn't matter...

teefixx, 2023-03-09:
is this done automatically? or do I need to have a special node for that?

teefixx, 2023-04-11:
> As a workaround, I set the Name in the incl des NodeRed at startup (over mqtt)

@pos-ei-don 

With which node do you set the battery name and how?

pos-ei-don, 2023-04-12:
@teefixx 

Yes, MQTT-OUT. Just write the name you wish at startup to the topic "W/[...]/CustomName" as string.
As the (at least the non ble driver) did not sent the custom field text you can enter in the BLE-APP and save on the bms, 
I used other values to "indentify" a BMS. Like Capacity. One CellPack got 280Ah, the other 281Ah.
If someone finds a better method please share.

So if is subscribe via mqtt-in to ""N/[vrm-id]/battery/+/InstalledCapacity", I get all my batteries in a seperate payload-message.
I then can easyle set the name as shown above.

teefixx, 2023-04-12:
@pos-ei-don 
Thank you
Tried this, changed the capacity in one battery in the BMS. This value is not read out, instead VictronGX takes the value from the serial driver...
What kind of BMS do you have? I have a Daly 16S

closed 2023-04-14 by mr-manuel

teefixx, 2023-04-21:
@pos-ei-don
sorry for bothering you again.
I found a way to distinct the batteries and write the values via MQTT out.
When I check it with MQTT Explorer the values are there.

[image]

Still, if I have a look at the remote console of VenusOS, the names did not change. They are still "SerialBattery (Daly)"

I tried both, publish at N and W...

pos-ei-don, 2023-04-21:
VenusOS requires a json object with value.
{value: "newName" }

teefixx, 2023-04-21:
that worked, thank you!