Louisvdw/dbus-serialbattery #445: Charger / Multiplus do not pay attention to Charge current limit

hoschult, 2023-01-31:
**Describe the bug**
- Charge current limit is set according to SoC to 10Amps and is reported correctly by vrm (see screenshot)
- However battery gets charged beyond this limit of 10 Amps by the victron RS450/100

**To Reproduce**
- Need to have an AC and DC coupled victron system with JBD bms --> see screenshot of plant topology
- Set DVCC to YES and use shared voltage / current sense
- Use serialbattery and the algorithm to reduce charge current in respect to SoC (Limit 10Amps when SoC >= 97%)
- In ESS, switch on AC and DC coupled PV feed in excess
- Let RS450/100 set under external control under BMS
- finally: Wait for a sunny day and observe that all power which is delivered from RS450 will be feed into battery (regardless charge current limit)

**Expected behavior**
I would expect that Venus will respect the charge current limit of JBD bms settings via dbms-serialbattery so that multiplus inverts all the power above 10Amps (coming from RS450) towards ACIn to grid for feed in

**Screenshots**

Overview about topology (at 12.19) where where charge current of 51.4Amps can be observed consumed by battery:

[image]

Charge current limit in VRM which was rightfully adjusted in respect to SoC:

[image]

Current flow chart towards battery:

[image]

Grid feed in settings of ESS:

[image]

Networked options of Victron Charger (RS450/100):

[image]

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi
 - Firmware Version: v2.92

** Bus-serialbattery is installed at 0.141

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JBD
 - Cells: 16
 - Interface:
```
                      Battery --> UART --> Raspberry Pi (VenusOS) <-- VeDirect <-- RS450/100
                                                                                  ^
                                                                              VEBus
                                                                                  ^
                                                            3phase cluster Multiplus 5000
```

wormsman, 2023-01-31:
As far as I know the Charge Current Limit is ignored from the MPPT when you have an ESS system and also DC Feed in activated. The question is why the multiplus do not feed in enough.

hoschult, 2023-01-31:
Hi wormsman, thx for pointing this out.  

> "The question is why the multiplus do not feed in enough"

Yes, exactly this are also my thoughts.... By hunting for answers I tumbled about this victron community posts: https://community.victronenergy.com/questions/91148/dvcc-limit-charge-current-not-working-with-grid-fe.html

I'm not quite sure in which direction to continue to search because it might be also an issue/bug which comes directly out of victron's ESS algorithm...

hoschult, 2023-02-09:
Hi all,

my system is now working with dc feed in and is respecting charge limits (CCL). The solution was to reduce voltages in utils.pl to:

```
MAX_CELL_VOLTAGE = 3.42                                                                                                
FLOAT_CELL_VOLTAGE = 3.37  
```

I think to lower the voltage of the DC-Charger (RS450) let the Multiplus to be able to transport current towards AC since battery not catches all the power (due to voltage spread charge voltage / battery voltage):

[image]

closed 2023-02-09 by hoschult

LarsAndersenDenmark, 2023-03-01:
Hi
Lowerig only helps stop charging earlier. The CCL is as far as I have experienced not accepted by what you are doing.
Could some coding in the Dbus battery help controlling the charge level some how? Maybe by setting a dynamic CVL based on the actual charge current seen, since the MPPTs still respect the CVL.