Louisvdw/dbus-serialbattery #450: BMS values are not shown correctly in log messages during startup

hoschult, 2023-02-06:
**Describe the bug**
During startup of serialbattery the values printed seems not to be correct. However, viewing cell voltage level via console seems to be right

**To Reproduce**
Use Jbd bms and connect via UART to raspberry

**Expected behavior**
right values to be shown - not sure if this is a general issue or only wrongly stated in log file

**Screenshots**

[image]

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi
 - Firmware Version 2.90 Venus os / serialbattery 0.141

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JBD
 - Cells: 16
 - Interface: See above

mr-manuel, 2023-05-02:
@iLeeeZi could you check, if you have the same behaviour with the nightly build?

iLeeeZi, 2023-05-02:
```
2023-05-01 16:11:40.291232500 INFO:SerialBattery:Starting dbus-serialbattery
2023-05-01 16:11:40.292203500 INFO:SerialBattery:dbus-serialbattery v1.0.0-jkbms_ble (20230501)
2023-05-01 16:11:40.292652500 INFO:SerialBattery:Testing LltJbd
2023-05-01 16:11:40.454924500 INFO:SerialBattery:Connection established to LltJbd
2023-05-01 16:11:40.455442500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyUSB3
2023-05-01 16:11:40.455913500 INFO:SerialBattery:========== Settings ==========
2023-05-01 16:11:40.456426500 INFO:SerialBattery:> Connection voltage: NoneV | Current: NoneA | SoC: None%
2023-05-01 16:11:40.456909500 INFO:SerialBattery:> Cell count: None | Cells populated: 0
2023-05-01 16:11:40.457395500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-05-01 16:11:40.457938500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 75.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A
2023-05-01 16:11:40.458411500 INFO:SerialBattery:> CVCM:     True
2023-05-01 16:11:40.458974500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.5V
2023-05-01 16:11:40.459685500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2023-05-01 16:11:40.460210500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-05-01 16:11:40.460751500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2023-05-01 16:11:40.506500500 INFO:SerialBattery:DeviceInstance = 3
2023-05-01 16:11:40.507177500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB3
2023-05-01 16:11:40.666449500 INFO:SerialBattery:publish config values = 0
``` 
Seems to be still the same

mr-manuel, 2023-05-02:
Thanks, I will check it

ramack, 2023-05-02:
While working on the Heltec BMS I currently see the same and I guess the output is written to the log, before the data is read by the corresponding methods.

mr-manuel, 2023-05-02:
Later today I release a fix. This comes from, that some BMS already read the data with `def test_connection(self):` and others not.

mr-manuel, 2023-05-02:
This is fixed with the latest `jkbms_ble` nightly for all BMS. Since I could not test all BMS, please leave a comment with your BMS name, if it's not fixed for you.

closed 2023-05-02 by mr-manuel

iLeeeZi, 2023-05-02:
```
2023-05-02 14:33:28.460361500 INFO:SerialBattery:========== Settings ==========
2023-05-02 14:33:28.460874500 INFO:SerialBattery:> Connection voltage: 13.18V | Current: -2.12A | SoC: 55.893382352941174%
2023-05-02 14:33:28.461542500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
2023-05-02 14:33:28.462006500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-05-02 14:33:28.462489500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 75.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A
2023-05-02 14:33:28.463013500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: NoneA | MAX BATTERY DISCHARGE CURRENT: NoneA (read from BMS)
``` 
Works with JBD. SOC might need some rounding :smile:

mr-manuel, 2023-05-02:
Ok, I added it. Will be in the next PR.

iLeeeZi, 2023-05-03:
@mr-manuel 
Can you change this:
https://github.com/Louisvdw/dbus-serialbattery/blob/jkbms_ble/etc/dbus-serialbattery/bms/lltjbd.py#L161

To
```
self.soc = round(100 * capacity_remain / capacity, 1)
``` 
Or
```
self.soc = round(100 * capacity_remain / capacity, 2)
``` 
To not lose improved resolution from #439

mr-manuel, 2023-05-03:
Sure

ramack, 2023-05-03:
Wouldn't it be best to round only on the output printing?