Louisvdw/dbus-serialbattery #458: Commit #393: BMS-own blocking charge/discharge not available

Dr-Gigavolt, 2023-02-12:
Hi Louis,
since the commit #393 the paths '/System/NrOfModulesBlockingCharge' and '/System/NrOfModulesBlockingDischarge' are always set by your code, regardless whether CCCM_CV_ENABLE and DCCM_CV_ENABLE are set True or False. It is not possible to transmit the state of MOSFET switch any more. Is there some reason for it? Otherwise please make the possibility to pass the BMS state again. Thanks.
Best regards,
Anton

Dr-Gigavolt, 2023-04-23:
Hi @mr-manuel 
I just need the option to get raw BMS data from Dbus as it was up to commit #393. Since 393 even if everything is set to False the Blocking charge/discharge values are from the Serial Battery driver and not from BMS. If the option "own charge parameters" in AggregateBatteries is activated, the calculation is interfered and therefore both drivers cannot be compatible any more. I didn't have the time to test jkbms_ble branch yet, I'll try it later.

mr-manuel, 2023-04-24:
Sorry, but I still can't understand.

The code from the `master` branch, `v0.14.3`, `v0.14.0` and `v0.13.0` is the same for the values `/System/NrOfModulesBlockingCharge` and `/System/NrOfModulesBlockingDischarge`.

## `master`
https://github.com/Louisvdw/dbus-serialbattery/blob/6471152bf701eb46beaa9ba4d11d93ffd1559dd4/etc/dbus-serialbattery/dbushelper.py#L374-L382

## `v0.14.3`
https://github.com/Louisvdw/dbus-serialbattery/blob/3685d15a5d1ba801867bbdb98f57fd805ea2c24a/etc/dbus-serialbattery/dbushelper.py#L244-L247

## `v0.14.0`
https://github.com/Louisvdw/dbus-serialbattery/blob/7aab4c850a5c8d9c205efefc155fe62bb527da8e/etc/dbus-serialbattery/dbushelper.py#L244-L247

## `v0.13.0`
https://github.com/Louisvdw/dbus-serialbattery/blob/6393e7698df274c8eee159dbd4d1ce909e5da74c/etc/dbus-serialbattery/dbushelper.py#L211-L214

However between `v0.13.0` and `v0.14.0` it was changed how `control_allow_charge` is set: [Comparing v0.13...v0.14 · Louisvdw/dbus-serialbattery](https://github.com/Louisvdw/dbus-serialbattery/compare/v0.13...Louisvdw:dbus-serialbattery:v0.14?expand=1#diff-9d9b25b4a5dcc9b6f162c96264601c1986789dd796a6b170586076a746fabbc0R157-R172). But this was one month before the commit you are referring to.

Am I missing something?

Dr-Gigavolt, 2023-05-04:
Hi @mr-manuel, I am sorry, I had no time yet.

Dr-Gigavolt, 2023-05-06:
Hi @mr-manuel , I've just installed the jkbms_ble and will observe the behavior tomorrow. I appreciate the well commented config.default.ini. But please explain me why the lead-acid algorithm bulk-absorption-float is still maintained? This art of choreography was developed for lead-acid cells having no balancer, no BMS and high leakage current.

mr-manuel, 2023-05-06:
There are only two voltages:
* Bulk/Absorption: For `Linear Mode` max voltage is applied the first time the driver runs. When the cells are balances, then max voltage is only applied again, if the cells are unbalanced in order to balance them.
* Float: Don't keep max voltage to reduce stress to the battery chemistry. This is based on this chart:
   [image]
   Source: https://www.sciencedirect.com/science/article/pii/S2352152X21015243

Dr-Gigavolt, 2023-05-07:
Hi @mr-manuel,

- thanks for the paper, I'll read it in following days. 
- I hadn't the issue with blocking charge/discharge when using jkbms_ble any more. But this driver is not stable, both instances disappeared several times per day.
- today I re-installed the driver at the time of a lot of solar power and full batteries. At the time it was not present the ESS tried to charge with couple of 10A. Do you know how to set up the ESS to disable charging and discharging if the battery monitor is not available? Thanks.

mr-manuel, 2023-05-08:
Which BMS do you have? Maybe Daly? If yes we imrpoved the driver a lot and now it runs finally smoothly. Would be great, if you can test it. It was merged today with the `jkbms_ble` branch of Louisvdw.

If the battery monitor is not available, then it will fall back to the default for a MultiPlus. There is no way to block discharge/charge, but you can prevent bigger currents by setting this settings to `0`. Don't forget to remove them afterwards.

GUI -> Settings -> DVCC -> Limit charge current
GUI -> Settings -> ESS -> Limit inverter power

transistorgit, 2023-05-16:
@Dr-Gigavolt can you check /var/log/messages.* for "watchdog" and "load average too high"? I've seen that before and wonder if it is the same

Dr-Gigavolt, 2023-05-16:
@mr-manuel 
> Which BMS do you have? Maybe Daly?

No, I have JK.

Dr-Gigavolt, 2023-05-16:
@transistorgit 
> @Dr-Gigavolt can you check /var/log/messages.* for "watchdog" and "load average too high"? I've seen that before and wonder if it is the same

I installed the jkbms-ble again to see the messages after the next crash.

Dr-Gigavolt, 2023-05-16:
The first crash already happened:
May 16 19:51:01 raspberrypi2 user.err kernel: [8031915.992387] usb 1-1.1.2.1: failed to send control message: -71
(no "watchdog" or "load average too high")

mr-manuel, 2023-05-18:
Could you post a screenshot of the result from the top command? Please watch for a minute how it's ongoing, so you get a feeling, when the best moment for a screenshot is. Else it can happen that you take a screenshot where the system is idle for a second.

I had a similar problem, when the device run out of memory. Then it bebun to swap and the CPU load went to the sky. This was caused by Node-RED.

Dr-Gigavolt, 2023-05-19:
I don't understand what you mean. The crash happens after some hours, I cannot sit and wait for it. But this commit works reliably and I use it since January: https://github.com/Louisvdw/dbus-serialbattery/tree/3685d15a5d1ba801867bbdb98f57fd805ea2c24a/etc/dbus-serialbattery

mr-manuel, 2023-05-20:
Reboot your Venus OS and check after 4 hours for one minute the top command. Then check again in 4 hours, if the system did not reboot. Post the screenshot here, after you understood which process is taking the most memory and/or cpu.

Dr-Gigavolt, 2023-05-21:
What is the "top command"?

Rikkert-RS, 2023-05-21:
@Dr-Gigavolt I think this is what is meant "top Command" https://www.linux-praxis.de/top-zeigt-laufende-cpu-prozesse

Dr-Gigavolt, 2023-05-27:
Hello @mr-manuel,
today I mounted jkbms-ble again. Before I updated the Venus, selected the small image (had large before). The instance on TTY3 crashed apx. 3/4 hour after Venus reboot. I found it out 13 min. later. The screenshots of processes are attached. I see nothing suspicious. 

[top_after_restart.txt](https://github.com/Louisvdw/dbus-serialbattery/files/11583077/top_after_restart.txt)
[top_13minutes_after_crash.txt](https://github.com/Louisvdw/dbus-serialbattery/files/11583078/top_13minutes_after_crash.txt)

mr-manuel, 2023-05-28:
I close this issue, since the issue related problem is solved.

@Dr-Gigavolt there is another issue open related to your current problem https://github.com/Louisvdw/dbus-serialbattery/issues/655

closed 2023-05-28 by mr-manuel