Louisvdw/dbus-serialbattery #464: Performance Problems with > than 6 Batteries, exclude some

pos-ei-don, 2023-02-16:
My Raspberry 4 8gb gets very slow when
A lot of batterys are connected.
As I dont need All batteries on its own, but need their data
For a additional Aggregated battery (which I use per mqtt), I suggest excluding some or All batteries from showing in the console, but deliver their data for use in ess.

Maxwell27321, 2023-03-08:
I have the same issue wie 4 banks, jk bms, connected to Cerbo GX via USB hub.

transistorgit, 2023-03-13:
which bms types and which interface? I found a bug relating to Daly and RS485 in that combination that would lead to slow updating.

pos-ei-don, 2023-03-13:
I read your progress with daly in the comments!
Unfortunately, I only use JK-BMS as they work best for me according to my tests.
I have a spare daly aswell, but don't like their balancing and i had the daly RS485 adapter getting hot and melting
which was no fun ;-)

mr-manuel, 2023-04-20:
@pos-ei-don could you check what causes the load? What is the output of this command?

```bash
top -b -n 1 | head -n 23
```

Best would be, if you keep watching `top` for a few seconds to understand, if there are spikes that cause the problem.

pos-ei-don, 2023-04-20:
Unfortunately, I cannot. I seperate the production-venusos  and one for testing. All batteries are not connected anymore, I only use your mqtt-battery on the production system for limiting Charge speed when the batteries are >55 Volts. And, I am not in the Country vurrently, so i cannot reconnect.
I can only say that my venusos is much more performant since then, and it is better in calculating settings. Also the gui is much faster.

mr-manuel, 2023-04-20:
Would you investigate further when you are back or can I close this issue?

closed 2023-04-27 by mr-manuel