Louisvdw/dbus-serialbattery #470: Charge and Discharge flickering (always jumps to 0) since driver is installed

MartinKutz, 2023-02-22:
**Describe the bug**
I had the Multiplus 2 running a week without connected BMS.
The charge and discharge behavior curve is in diagram 1.

Yesterday I installed the dbus-serialbattery driver and I can see all the Daly BMS values in VenusOS. But charge and discharge jumps always to 0.

**Screenshots**
[1] Charge/Discharge before:
[image]

[2] Charge/Discharge after:
[image]

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi 4
 - Firmware Version: 2.92

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: Daly BMS, LifePo4 Battery
 - Cells: 16
 - Interface: Uart

mr-manuel, 2023-03-12:
Have you enabled or disabled this?

https://github.com/Louisvdw/dbus-serialbattery/blob/3685d15a5d1ba801867bbdb98f57fd805ea2c24a/etc/dbus-serialbattery/utils.py#L35-L38

Can you try what happens with `v0.14.2`?

MartinKutz, 2023-03-15:
Thanks for your response!
LINEAR_LIMITATION_ENABLE is set to false. What is the recommendation to this setting?

Ok, I will try v0.14.2 in the next few days.

mr-manuel, 2023-03-15:
Unfortunately I have no reccomendation for this setting, maybe someone other.

MartinKutz, 2023-04-21:
@mr-manuel sorry, I had no time last weeks. I will have a try beginning of next week.

mr-manuel, 2023-04-21:
Maybe also this can help you https://github.com/Louisvdw/dbus-serialbattery/issues/371.

MartinKutz, 2023-05-02:
Today I had time to test it again.

I cannot get Version v0.14.2 to work.
This is my modified install script:
`
#!/bin/bash
set -x
curl -s https://api.github.com/repos/Louisvdw/dbus-serialbattery/releases/86212497 | grep "browser_download_url.*gz" | cut -d : -f 2,3 | tr -d \" | wget -O venus-data.tar.gz -qi -
tar -zxf ./venus-data.tar.gz -C /data
sh /data/etc/dbus-serialbattery/reinstalllocal.sh
`

With current version it's running.

Issue #371 has the flickering behavior only in the near of the cut of voltage - I have the flickering all the time at charging and discharging.

I reinstalled the newest version and for some reason I havn't the behavior anymore.
I will monitor it for the next days. But seems to work.

[image]

Thanks for your help!

mr-manuel, 2023-05-02:
With the latest code we changed a lot and fixed a lot of issues. Feel free to reopen the issue, if you see the same error again.

closed 2023-05-02 by mr-manuel