Louisvdw/dbus-serialbattery #471: MP keeps charging and discharging at set point MAX_CELL_VOLTAGE

Joeknx, 2023-02-22:
Hello,

my Multiplus II keeps changing between charging and discharging at MAX_CELL_VOLTAGE = 3.45 (55.2)
Every 10 sec or so. 

Is this a normal behavior, or did I miss any settings?
Versions:
0.14.3
Venus 3.00-15

Thank you!

mr-manuel, 2023-02-22:
Something like that? https://community.victronenergy.com/questions/1976/ess-short-discharge-pulses-in-float.html

Joeknx, 2023-02-22:
It doesn't say Float, but the problem seems the same. Maybe it is normal to keep the 55,2V

mr-manuel, 2023-02-22:
In the OverView it doesn't say float? Have you checked the "Device List"?

[image]

[image]

Joeknx, 2023-02-22:
The device list looks like yours and says absorption. 
The Amps from "SerialBattery(Jkbms)" flickers between "-" and "+" and in the VRM it changes between "Discharging" an "External control"

Joeknx, 2023-02-22:
I'm not sure if you mean this:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

pos-ei-don, 2023-02-22:
I have Float Set to 56.8V, Absorbtion to 57.8v, but stop charging using the serialBattery-settings at 55.2V, like as you did directly on the MPII.
And I set the Max charging voltage to 56.3v.
Seems wo work better in this case, but maybe settings can be improvend further.

Joeknx, 2023-02-22:
Ok thank you. 
Maybe the MPII  and serialbattery are interfering. 
I try to change the Settings in the MPII tomorrow

morres83, 2023-02-27:
I would also like to follow up on this topic, I think I have a similar issue.
I have deactivated current limits based on SoC (@Joeknx is it  the same for you?) because I think that the SoC from BMS is getting massively shifted when the battery is not being fully charged for about 12-14 days (currently not enough sun, the SoC is floating between 15-50% for the last 2 weeks.). Actually it says 15% but the voltage is so low that it more or less corresponds to 1-5%. This error also propagates to the full charging, meaning SoC is shown as 100% (I tested by charging from grid) but the voltage is too low (<3.4V). That was the reason that I deactivated limits on SoC and relied on charging limits based on voltage.

Now what happens is that it charged to 100% SoC (after the above mentioned 2 weeks of being betw. 10-50%) but the cell voltages are still so low that it keeps charging with 60A for 1-2 hours. I think it could be okay but the issue is that when one cell hits the limit (e.g. 3.45 V) the first current limit applies 60A-30A. 
-> Immediately the cell voltages drop (being below 3.45V)
-> The full current gets released again (30A->60A)
-> The cell voltages grow fast and hit the limit again , and so on
It seems being in a loop and I guess it is not how it should be.

The problem of this topic here sounds very similar.
So I fear that the charge controlling has to rely on SoC and this setting must not be turned off.

Anyone with a suggestion on it?

Joeknx, 2023-02-27:
> The problem of this topic here sounds very similar. So I fear that the charge controlling has to rely on SoC and this setting must not be turned off.
> 
> Anyone with a suggestion on it?

Yes, @morres83 I also have deactivated charging based on SOC.

But my "problem" is different. Only the inverter changes pretty quick between charging and discharging but keeps 55.2V
My guess is, the this is the normal behavior to keep the limit!?

Are your Cells balanced? Sound like they are not.

[image]

morres83, 2023-02-27:
Yes my cells are balanced so this shouldn't be the problem. For me it seems that when the current drops, also the voltage drops. Maybe I have to wait a longer time until it settles, but today it wasn't possible because the sun was gone.

For reaching your 55.2V, did you charge a long time when it was showing already 100%?

Joeknx, 2023-02-27:
It didn't take a long time. 
What I know is that when the Voltage drops when the current goes down, the Cells are not full!
So you probably have to wait a longer time.

What I noticed is that serial battery takes the MPII Voltage to limit and not the BMS Voltage. I have selected the BMS as battery monitor!
The Voltages were different so I saw it. Now I calibrated the BMS and it does not matter anymore.

morres83, 2023-02-27:
I have also selected JKBMS as battery monitor, but it is an interesting point about the voltages. So you are saying you have calibrated the JK BMS to "see" the same voltage as the MP II? Currently they also differ for me (a bit):

[image]

[image]

pos-ei-don, 2023-02-27:
... But this Dr er can throttle the charging amps linear to cell voltage. So the should never be this on/off All the time, it should narrow the soc  to 100% slower and slower.

morres83, 2023-02-27:
Hello @pos-ei-don , yes now when you are saying this, normally it should be linear. But I am very sure that there was never a linear interpolated value. The parameter either showed 60A or 30A (so current at 3.3V point or current at 3.45V point) and nothing in between. So I was digging around and I found that LINEAR_LIMITATION_ENABLE in involved here. I will try with this option turned on to get rid of the jumps.

What is CVCM_ENABLE, I am not sure about this one...

pos-ei-don, 2023-02-27:
Sorry, nö. Unfortunately it seems that I mixed it up with an fork I once tried. So sorry, I cant help currently

Bonzai10, 2023-03-01:
charging/discharging when the battery is full is normal. the driver does not go into float mode under linear.

test the version 0.14 version. This works for me, I think 0.14.3 has an error in this regard.
Values are taken from the SOC calculation although linear is set.

seidler2547, 2023-03-16:
I think one of the things that would be important here is to reduce the current earlier. I set `LINEAR_LIMITATION_ENABLE` to `True` and I reduced the voltages in this table:
```
# Set Steps to reduce battery current. The current will be changed linear between those steps
CELL_VOLTAGES_WHILE_CHARGING         = [3.55, 3.48, 3.42, 3.30]
MAX_CHARGE_CURRENT_CV                = [   0,    2,  40,  80]
```
(Note that I want 80A max charging.) This way, the charging current is reduced early and there is a nice lowering charging curve towards the top:
[image]

In the rightmost graph, the orange line is the current limit imposed by the driver. You can see that even though it was 100% sunny today, the MPPT started throttling the charge as the battery got more full.

mr-manuel, 2023-04-14:
> charging/discharging when the battery is full is normal. the driver does not go into float mode under linear.
> 
> test the version 0.14 . This works for me, I think 0.14.3 has an error in this regard. Values are taken from the SOC calculation although linear is set.
> 
> see here: #421

Is solved with the next version.

closed 2023-04-14 by mr-manuel