Louisvdw/dbus-serialbattery #473: Balancing of cells not clear when using Daly BMS

mrgithub, 2023-02-22:
Is there a way to log the individual battery cell voltages over time into a csv file to analyse the performance of individual cells? Or even better, log it in a VRM graph?

The cells should turn red when they are balancing but it seems all the cells are red all the time in the console. Can the min and max cells be a different colour and can only the cells being balanced be highlighted?

mr-manuel, 2023-02-22:
There are many ways to log individual cell voltage over time, but there is no out of the box solution as I know.

In my eyes a CSV it not appropriate for this, since it would wear the SD card much faster. All cell voltages are published over MQTT. You can fetch this data with Node-RED (included in the Venus OS Large image) and send it to a InfluxDB. Once in the InfluxDB you can display it with Grafana. You can run all products even as Docker containers on a separate Raspberry Pi, Computer or whatever you want. InfluxDB and Grafana have to run on another system and cannot run on Venus OS.

Here an example what it could look like:
[image]

If you still want to write to a CSV you can use Node-RED and write the data on the SD card.

Since the data is published over MQTT there are a lot more possibilities.

For the other I unfortunately cannot answer you.

sanderjilderts, 2023-03-08:
@mrgithub

I have the same problems the cells are red all the time. 

[image]

hansgeorgk, 2023-03-10:
> In my eyes a CSV it not appropriate for this, since it would wear the SD card much faster. All cell voltages are published
> over MQTT. You can fetch this data with Node-RED (included in the Venus OS [...]

So it would look like the following?
GX devices -> Node-RED -> InfluxDB -> Grafana
whereas Node-RED, InfluxDB, Grafana are all running as docker containers on a RaspberryPi?
or do I need to get Node-RED (Venus Large) installed on my GX device? If that is possible/recommended at all ...

thanks
cheers
HG

mr-manuel, 2023-03-15:
Here you can find a Node-RED example flow and Grafana dashboard: https://github.com/Louisvdw/dbus-serialbattery/discussions/495#discussioncomment-5318939

closed 2023-04-14 by mr-manuel