Louisvdw/dbus-serialbattery #487: Multiplus II affected: max charging current 10 A, SOC missing

Maxwell27321, 2023-03-09:
My installation:
- 3 phase Multiplus II 5000/48: Firmware 500
- 4 battery banks (each LiFePo4 with eve 280Ah) with JK BMS : B2A24SP15-CAN
- Cerbo GX: VenusOS 2.93  
- Lynx Shunt 
- MPPT RS 450/200

Motivation to install the BMS driver: 
- get alarms if the batteries are not working well, 
  but the BMS should be just act as an measuring device. 
- Lynx shunt sees the sum of current and the overall SOC

I followed the very simple installation (thanks) and did not see any issue

After installing the BMS driver:
- All 4 BMS show up, and I can access all information, but …
[image]

- charging current of the Multiplus is limited uncontrollable to 10A by something, 
  regardless the settings in the driver (CCL) or the DVCC (or ESS) in the Cerbo.
[image]

- Communication to Multiplus seems to be affected, 
   as SOC is not showing up any more.
[image]

- disabling the driver leads to „Low Battery Voltage Alarm“ from the Multiplus, 
  and the charging current is still limited to 10 A, even with the DVCC is set to 100A. 
[image]

- in the VRM, I see 5 JKs and not 4, as installed 
[image]

Q: Can I debug the system to see what is going wrong here? 
Q: Is there a way to complete remove the driver?
Q: Is the only way to get the charging current of DVCC is to reinstall VenusOS?

Thanks for any help

mr-manuel, 2023-03-09:
Hello,

sure you can debug. Check out the wiki https://github.com/Louisvdw/dbus-serialbattery/wiki/Troubleshoot

In the wiki you find also how to disable the driver https://github.com/Louisvdw/dbus-serialbattery/wiki/How-to-install#disable-the-driver

1. Did you set the 100 A Limit in DVCC?
2. What do you see under `SerialBattery -> Parameters` on each battery?

Maxwell27321, 2023-03-09:
Hi. 

1. Yes, 100 A limit in DVCC
2. [s.b.](url) the MPPT is kicking in with 1000W, so I got higher currents no

Set per util-file is: CCL 50A and DCL 60 A

Voltages are all 53.62 - 53.64 V

Bank 1: CCL 50 A, DCL 60 A SOC 90%
Bank 2; CCL  5 A,  DCL 60 A, SOC 98% <- SOC is not correctly estimated from the BMS
Bank 3: CCL 50 A, DCL 60 A, SOC 89%
Bank 4: CCL 25 A, DCL 60 A, SOC 94%

That looks for me, that the method which reduces the current is working. 

BTW: I see higher currents than 5 A in Bank 2. Could be that I set CCL used to False. I‘ll check later.

pos-ei-don, 2023-03-09:
... And I think the multiplus will use Max Charge current of the lowest battery. I had this issue some days aswell.
Tjlhats why I whish to hide All Single batterys from System and want to only present a Aggregated battery with multiple "modules".

teefixx, 2023-03-09:
you can do the aggregated battery with 
https://github.com/arrow1800/Victron-VirtualBattery

works quite well and you can adjust all parameteres in node red...

mr-manuel, 2023-03-09:
> you can do the aggregated battery with https://github.com/arrow1800/Victron-VirtualBattery
> 
> works quite well and you can adjust all parameteres in node red...

Thats a nice idea!

If you prefer to feed a virtual (aggregated) battery over MQTT you can also use the driver: [dbus-mqtt-battery](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery)

You can also get the data via Node-RED from Venus OS and then send it to a topic you decided in the `config.ini`. If there is a value missing that you need, I can add it to the driver.

pos-ei-don, 2023-03-09:
Great New rldriver, will give it a try!
I just need a Parameter non the serialBattery driver to only sent the data to the mqtt, without registerin it as a stand alone battery in VenusOS.

teefixx, 2023-03-09:
I have all three batteries in venusOS: 2 real and one virtual. The virtual is assigned in ESS to be used, the other 2 are just info...

mr-manuel, 2023-03-09:
> Great New rldriver, will give it a try! I just need a Parameter non the serialBattery driver to only sent the data to the mqtt, without registerin it as a stand alone battery in VenusOS.

Right now the driver can't send the data over MQTT without registering as battery, but you can try to select the `MQTT battery` as battery monitor under `Venus OS GUI -> Settings -> System setup ->Battery monitor `.

I don't know, if this direction (to only send data without registering) is intended, since this would complicate thing unnecessarly in my eyes.

pos-ei-don, 2023-03-09:
Of course I did so. But there are some Problems with this:
1) venusos gets really slow when 3 or more batterys are connected
2) venusos seems to select the lowest charging current, regardless which device is selected there

mr-manuel, 2023-03-09:
1. What does the CPU load say? Could you check with the `top` command?
2. That would make sense, since all batteries are sharing the current which is limited by the charger/inverter.
You can try to disable all current control management options from the serialbattery, BUT pay attention. In this case you have to manage this by yourself with Node-RED and the virtual battery driver. In the worst case, the BMS should interact.
In `utils.py` set all variables to `False` that end with `_ENABLE`

Just to remember: When using multiple batteries in parallell, you can't tell which battery should charge with which current.

pos-ei-don, 2023-03-09:
1) I have to change it back to test. Currently I have a secondary rpi to send the battery data to mqtt.
2) ok, I test this on weeckend. I can charge over 1000A, having several batterys in parallel, so this is not dangerous in my current setup. But that is why I dont care if a Single cell Blocks charging.

mr-manuel, 2023-03-09:
@pos-ei-don your setup sounds interesting. Do you have some more details? :)

Bonzai10, 2023-03-09:
as this bug is not fixed until now, you have to change to v14.0
https://github.com/Louisvdw/dbus-serialbattery/issues/421

in 14.3 Charge current control management referring to cell-voltage is not running

Maxwell27321, 2023-03-09:
All.

Thx for comments so far. I have all current control disables in utils.py.
However the I do not get higher currents from my 3 x Multiplus 5000/48/70.

     3.5 A per battery bank. 

So where do the Multiplus get this current limitation from? They should use the DVCC, right?

And, I still do not see a SOC in the tab of the Multiplus, as shown above.

Maxwell27321, 2023-03-09:
Ok, I solved it. The MultiPlus need to have a restart. Now charge current from the Multiplus is controlled via DVCC settings.

The virtual battery is interesting to me. Are all individual voltages of cells listed here as well?
Can I set alarms for the virtual battery?

mr-manuel, 2023-03-09:
The virtual battery doesn't list all cell voltages. Check the full [JSON structure](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery#json-structure) to see which values you can feed. I think it would not make many sense, if you see already all batteries in the GUI and then a battery again the same data. Maybe do you have an example to understand the use case?

The driver [dbus-mqtt-battery](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery) has following alarms at the moment, based on the configuration from the `config.ini`:
* LowVoltage
* HighVoltage
* LowSoc

In the [dbus documentation](https://github.com/victronenergy/venus/wiki/dbus#battery) you can find the possible alarms.

I can add other alarms that can be triggered over MQTT, if you need them.

mr-manuel, 2023-03-10:
I just released a complete new version of the [dbus-mqtt-battery](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery) driver. Now it supports also cell voltages and all warnings can be set over MQTT.

Bonzai10, 2023-03-12:
@ mr-manuel
Hello,
has the error from #421 been repaired in your version of the driver?
I think Louis doesn't have that much time to keep the driver up to date at the moment.
I'm currently running Waldemarfrech 0.14.3 because it works. but would love to have all the new ads from you too :-)
Does the Bluetooth version also now run without crashes?

mr-manuel, 2023-03-12:
@Bonzai10 no, I did not address https://github.com/Louisvdw/dbus-serialbattery/issues/421 yet.

I will check the differences to Waldemarfrech 0.14.3.

The BLE driver in my repo has implemented an additional workaround to prevent problems. It restarts the Bluetooth service every 12 hours via cronjob.

mr-manuel, 2023-03-12:
@Bonzai10 please check https://github.com/Louisvdw/dbus-serialbattery/issues/421 to test the fix.

Bonzai10, 2023-03-13:
Hello,

Thanks. i will test it tonight. can you offer a complete download of your installation? or do I have to install 14.3 from Louis first and then copy your files into it?

Sorry, I'm not a hero in this regard :-(

mr-manuel, 2023-03-13:
You can just follow https://github.com/Louisvdw/dbus-serialbattery/issues/421#issuecomment-1465329297. It downloads the whole driver from the branch of my fork and install it.

mr-manuel, 2023-04-09:
@Maxwell27321 was your issue solved? If not which issues still persist?

closed 2023-04-20 by mr-manuel