Louisvdw/dbus-serialbattery #494: Multiplus II GX: changes in utils.py have no effect

SolarGuy724, 2023-03-13:
No matter how hard I try: I am not able to change charging current on my GX (Jiabaida BMS). 

Latest release dbus-serialbattery. 
dbus-serialbattery is installed and seems to work. All Venus menus as expected.
Limits are standard dbus-serialbattery 60/50 A due to venus.

System is working......

[image]

[image]

My problem:
Changes in util.py have no effect.  

[image]

What I did:

- disable all 6 controls (i.e. in utils.py set all variables to False that end with _ENABLE). 
- changed in utils.py : CCL 100A and DCL 100A
- complete reinstall (clean-up and "fresh" install)
- reboot etc.

Can anyone please give me a hint what went wrong?

SolarGuy724, 2023-03-13:
Issue solved! 

The problem was the starting-order in rc.local AFTER the ET340 Faker. Just moved it before ET340 Faker and the system works just fine :-).
Changes in utils.py now have the desired effects!

closed 2023-03-13 by SolarGuy724

skruehoetl, 2023-04-24:
Could you maybe give a few more details on you solution? I'm facing a similar-feeling issue with the jkBMS attached to the CerboGX of a friend. We keep trying to modify the CC max in the utils.py - and nothing happens.

mr-manuel, 2023-04-24:
That makes no sense that the `/data/etc/dbus-serialbattery/utils.py` is overridden on each reboot.

Did you also check the charge current based on SoC?

mr-manuel, 2023-04-24:
Ah, maybe it has to do something with the readonly mounts. Try to execute this before changing the `utils.py`. But I never heard before that this affects data in `/data`.

https://github.com/Louisvdw/dbus-serialbattery/blob/3685d15a5d1ba801867bbdb98f57fd805ea2c24a/etc/dbus-serialbattery/reinstalllocal.sh#L5-L6

skruehoetl, 2023-04-24:
> That makes no sense that the `/data/etc/dbus-serialbattery/utils.py` is overridden on each reboot.
> 
> Did you also check the charge current based on SoC?

Yes. I just checked the current value in the HomeAssistant and at 90 % SoC the CCMax went down to 25 A (from 50 A - where I had set 100 A). I logged into the Cerbo and saw that `/data/etc/dbus-serialbattery/utils.py` is back to default again - so it contains settings for CCMax 50 A - so the 25 A match CC_CURRENT_LIMIT3.

skruehoetl, 2023-04-24:
> Ah, maybe it has to do something with the readonly mounts. Try to execute this before changing the `utils.py`. But I never heard before that this affects data in `/data`. [...]

Okay... sounds intrigueing. I gave it a try and will report back as soon as my friend greenlights me rebooting his Cerbo...

skruehoetl, 2023-04-25:
I have to apologize.

The problem is solved. I tried the remount-rw script before modifying the utils.py - and then I looked into the mounted volumes...
... and saw a SD-card in the list. Turns out, that we still had the SD-card from the original install in the slot and on each reboot, the driver happily kept overwriting itself.

Problem solved - and once more it only existed between keyboard and chair.

Sorry for bothering you and a big THANK YOU for your help!

mr-manuel, 2023-04-25:
I'm happy that you solved it. You're welcome :-)