Louisvdw/dbus-serialbattery #498: BMS turns off mppt at high power

gerd-git, 2023-03-19:
**Describe the bug**
BMS turns off mppt charge controller at high power
Charge controller switches from "Ext.Control" to "Off", then to "Wakeup" and then back to "Ext.Control" then back to "OFF" -> "Wakeup" -> "Ext.Control" -> "OFF" etc. This endless loop happens with high power from about 800W or higher.

**To Reproduce**
My system:
13 modules with 385Wp
MPPT 150/60 (East) : 3S => 1155 Wp
MPPT 150/70 (West): 2S2P => 1540Wp
MPPT 150/100 (South): 3S2P => 2310 Wp

**Expected behavior**
Charge controller remains in "Ext.Control" mode

**Screenshots**
I only saw this behavior with the west and south charge controller:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

**VenusOS :**
- Device type: Raspberry Pi
- Firmware Version: v2.93
- SerialBattery: 0.142

**Battery/BMS :**
- JK-BMS 
- Cells: 8
- Interface: USB-RS485

Thank you for help...

mr-manuel, 2023-04-24:
Hello,

did you troubleshoot? See [How to troubleshoot](https://louisvdw.github.io/dbus-serialbattery/troubleshoot/) for that.

1. What does the log say?
2. What happens, if you disconnect the BMS for testing? Is the issue then still present?

gerd-git, 2023-04-25:
You can close this bug. That was my operator error.

mr-manuel, 2023-04-25:
What was the problem/error? Maybe it also can help others.

gerd-git, 2023-04-26:
wiring error