Louisvdw/dbus-serialbattery #501: Integrate JK BMS or other via MQTT

ayasystems, 2023-03-27:
**Describe the solution you'd like**
I have a JK BMS and i read it over BLE with ESPHome. All BMS data ir published over MQTT. Could you add the posibility to read BMS data over MQTT with your driver?

pos-ei-don, 2023-03-27:
I would suggest using this:
https://github.com/freakent/dbus-mqtt-devices

Can you tell me more about how you connected the ESP-Home to the BMS?
I'm interested in this aswell!

mr-manuel, 2023-03-27:
Hi @ayasystems,
if you use the dbus-serialbattery driver all the BMS data is automatically published to the local MQTT broker from Venus OS. There is also a Bluetooth version for the JKBMS in development. See the link above from Louisvdw.

If you prefer to feed a virtual battery in Venus OS with MQTT data, then this might be useful for you: [venus-os_dbus-mqtt-battery](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery)

ayasystems, 2023-03-27:
@Louisvdw and @mr-manuel thank you very much!! i will review it!!

@pos-ei-don I'm using https://github.com/syssi/esphome-jk-bms to read the jk-bms with esphome

Regards!

closed 2023-04-05 by Louisvdw