Louisvdw/dbus-serialbattery #508: Add Support for Heltec BMS

ramack, 2023-04-04:
**Is your feature request related to a problem? Please describe.**
I read on https://github.com/Louisvdw/dbus-serialbattery/wiki/BMS-types-supported that "JKBMS / Heltec BMS" are supported, so I tried to connect my Heltec Smart BMS 9-25S 100A RS485 but it is not detected by the dbus-serialbattery driver.

**Describe the solution you'd like**
According the docs I have (and what I spied on the RS485 with the scope) the BMS seems to talk Modbus RTU and is different from the JKBMS. So I propose to add support for this Heltec device. I don't know about other Heltec BMSs and if there is a variant that is compatible to JKBMS, but we might also want to update the docs accordingly.

ramack, 2023-04-15:
I am working on this. @Louisvdw  @mr-manuel should I base that work directly on the JKBMS-BLE branch?

mr-manuel, 2023-04-16:
If you are not adding a BLE driver than you can merge with the master branch. The only thing is that there are a lot of changes outstanding, see https://github.com/Louisvdw/dbus-serialbattery/pull/468.
If you are adding only a new python file and the corresponding additions to include the file it should not be hard to merge.

ramack, 2023-04-16:
OK thanks. The config.ini seems to be a very interesting change there... maybe the timeline to merge that into master us the main question right now

mr-manuel, 2023-04-16:
Unfortunately @Louisvdw has not much time right now, but he will do so after 2nd May. If you want to keep it simple merge with the `jkbms_ble` branch, than I can review your PR and merge it directly.