Louisvdw/dbus-serialbattery #512: Unstable communication

Folho39, 2023-04-06:
Hi,
I have installed the driver for communication on my system earlier this week and am using this with a Uart port on my JBD BMS.
The Bms was recognized and looked to be working ok, however when I keep an eye on the meaured values from the bms, all signals drop away and come back. and each time this is happening, the inverter stops using power from the battery.

Any idea how I can get the communication more stable?
am using a multiplus 5000 48V with inbuild Gx device and the system runs as ESS

Thanks in advance,

for the communication I am using the uart converter provided by JBD, which i also use for the configuration from the module.

pos-ei-don, 2023-04-20:
I saw them unstablenusing cheap usb Hubs  or overpowered usb ports. The best solution for me was to use a "rpi usb hat" Board, with a good Power supply.

Folho39, 2023-04-22:
Hi,
Thanks for the reply, The cable between the usb adapter and the BMS was unshielded, as i could not find a shielded cable I have created a shield and this looks to be working, before until 750watt was working, above problems would occur, I have just tested with 2kw after placing the shield and this was without problems.so it looks to be the cause.

mr-manuel, 2023-04-22:
Could you post a picture of your installation? Maybe this help also others.

closed 2023-04-24 by mr-manuel