Louisvdw/dbus-serialbattery #514: Cell tradings of Daly BMS

DIVERSDIVERS, 2023-04-07:
Hi, 

i can not see single cells ins Venus OS.
(diver support for Daly BMS to track each cell)

My specifications:

BMS: Daly BMS 250/125 8S
BMS connection: via RS485 to USB
Cells: 8 cell configuration 
Diver Version: v 0.14.3
Venus OS Version: Color GX Venus OS v 2.93
Configuration: see attachment

[utils.zip](https://github.com/Louisvdw/dbus-serialbattery/files/11177249/utils.zip)

Tank you

rbeudel, 2023-04-07:
Look here [https://github.com/Louisvdw/dbus-serialbattery/issues/397](url)

DIVERSDIVERS, 2023-04-07:
i looked, thank you but where i have to do this changes?

rbeudel, 2023-04-07:
[daly.zip](https://github.com/Louisvdw/dbus-serialbattery/files/11177372/daly.zip) It is the daly.py file in the same folder like the utils.py. Backup your daly.py and try the linked one.

DIVERSDIVERS, 2023-04-07:
thank you very much it worked.

Last Question: is it possible to store the name, beacause after every reboot from Venus OS allways the same standard name "SerialBattery(Daly)"
[image]
[image]
[image]

mr-manuel, 2023-04-07:
It is not yet possible to store the name. It's planned for the future.

DIVERSDIVERS, 2023-04-07:
ok thanks for help for both

mr-manuel, 2023-04-07:
Can probably be closed when https://github.com/Louisvdw/dbus-serialbattery/pull/484 is merged.

mr-manuel, 2023-04-07:
If you want to close it already even better :)

closed 2023-04-07 by DIVERSDIVERS

teefixx, 2023-04-08:
> thank you very much it worked.
> 
> Last Question: is it possible to store the name, beacause after every reboot from Venus OS allways the same standard name "SerialBattery(Daly)" [image] [image] [image]

Hi

Can you tell me how you go rid of the fields you don't need, eg Bus voltage, Starter voltage, Top section voltage, ...
I see on your photos, that only the fields you need are displayed...

Thank you

mr-manuel, 2023-04-08:
Probably you have Venus OS > 3.00~14. There is already a PR https://github.com/Louisvdw/dbus-serialbattery/pull/468 open to solve this. You can manually install a copy of the master branch from my fork, if you want to get rid of until the PR is merged.

teefixx, 2023-04-09:
@mr-manuel 

I did that today, but after that the whole cell voltage block was missing…
Installed Venus Version: 3.00-21

mr-manuel, 2023-04-09:
Because you have to apply the Daly fix after (https://github.com/Louisvdw/dbus-serialbattery/pull/484). It's not applied in my fork.

teefixx, 2023-04-10:
This worked, thank you mr-manuel.
Do you plan to publish a release on github or will you stick to manual install?

mr-manuel, 2023-04-11:
No, you will have to wait for the open PR to be merged.