Louisvdw/dbus-serialbattery #516: Cell tradings of Daly BMS - Not updating

DIVERSDIVERS, 2023-04-07:
Sorry to bother you again, unfortunately it doesn't update the cell voltage
[image]
[image]

rbeudel, 2023-04-07:
[daly.neu.zip](https://github.com/Louisvdw/dbus-serialbattery/files/11179594/daly.neu.zip) I dont have changed all parameter like in #397
I did this and I hope it will be fine for you and I did not mistake. @towatai reported his 8s runs.

DIVERSDIVERS, 2023-04-07:
thank u i tried it still not updating single cells, only one when i reboot
[image]

rbeudel, 2023-04-07:
Then I cant help you ;-)

towatai, 2023-04-07:
What Type of Cable is used? A original Daly? i´ve switched back to the UART ---> USB Cable because the RS485 make Problems together with Aggregate Batterys as descriped in #397
I only use original Daly-Cables!

DIVERSDIVERS, 2023-04-08:
original Daly RS485 to USB

transistorgit, 2023-04-14:
Hi, the problem is because of bugs in the serial interface programming for daly devices. I fixed it in the pull request https://github.com/Louisvdw/dbus-serialbattery/pull/484
still waiting for approval, but you can try on your own.

transistorgit, 2023-04-14:
Duplicate of #397

closed 2023-04-14 by mr-manuel