Louisvdw/dbus-serialbattery #518: JKBMS V11.26

githubfrance, 2023-04-08:
Hi Louis,
Thank you for your work.

I have a new environnement with a 
JK-BMS V11.26 4 cells 
raspberry pi 2 
Venus os large (last release)
Apr  8 15:12:00 raspberrypi2 user.info kernel: [   68.231764] usb 1-1.3: new full-speed USB device number 7 using dwc_otg
Apr  8 15:12:00 raspberrypi2 user.info kernel: [   68.365159] usb 1-1.3: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
Apr  8 15:12:00 raspberrypi2 user.info kernel: [   68.373710] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
Apr  8 15:12:00 raspberrypi2 user.info kernel: [   68.381129] usb 1-1.3: Product: USB Serial
Apr  8 15:12:00 raspberrypi2 user.info kernel: [   68.387871] ch341 1-1.3:1.0: ch341-uart converter detected
Apr  8 15:12:00 raspberrypi2 user.info kernel: [   68.406012] usb 1-1.3: ch341-uart converter now attached to ttyUSB0
The converter is recgonized

root@raspberrypi2:~# tail -f /data/log/serial-starter/current
@400000006431851305d078d4 INFO: Start service gps-dbus.ttyUSB0 once
@400000006431851c1c6e8aa4 INFO: Start service vedirect-interface.ttyUSB0 once
@40000000643185212efebd24 INFO: Start service dbus-serialbattery.ttyUSB0 once
@400000006431852d0b78ab1c INFO: Start service gps-dbus.ttyUSB0 once
@400000006431853621d60a6c INFO: Start service vedirect-interface.ttyUSB0 once
@400000006431853b3428ca74 INFO: Start service dbus-serialbattery.ttyUSB0 once
@400000006431854710fbffe4 INFO: Start service gps-dbus.ttyUSB0 once
@400000006431855026fbc25c INFO: Start service vedirect-interface.ttyUSB0 once
@4000000064318555397e1de4 INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000064318561173e283c INFO: Start service gps-dbus.ttyUSB0 once

 tail -f /data/log/dbus-serialbattery.ttyUSB0/current
@40000000643185293a1911dc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000643185293a329904 INFO:SerialBattery:Testing Lifepower
@400000006431852a0eca75fc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431852a0edd84e4 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@400000006431852a0eeaaffc INFO:SerialBattery:Testing Renogy
@400000006431852a1f425f6c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431852a1f567fc4 INFO:SerialBattery:Testing Renogy
@400000006431852a2f916fd4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431852a2fa282ec INFO:SerialBattery:Testing Ecs
@400000006431852b1568760c ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
@400000006431853d1809bd6c INFO:SerialBattery:Starting dbus-serialbattery
@400000006431853d181db6b4 INFO:SerialBattery:dbus-serialbattery v0.14.3
@400000006431853d182ec9cc INFO:SerialBattery:Testing LltJbd
@400000006431853d28a0df0c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431853d28bdf074 INFO:SerialBattery:Testing Ant
@400000006431853d396af78c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431853d398063ec ERROR:SerialBattery:>>> ERROR: Incorrect Data
@400000006431853d398b683c INFO:SerialBattery:Testing Daly
@400000006431853e0e5bae9c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431853e0e766674 INFO:SerialBattery:Testing Daly
@400000006431853e1eb68a8c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431853e1ecd279c INFO:SerialBattery:Testing Jkbms
@400000006431853e2f165434 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431853e2f2d1084 INFO:SerialBattery:Testing Lifepower
@400000006431853f040b7fe4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431853f04223464 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@400000006431853f042c6d94 INFO:SerialBattery:Testing Renogy
@400000006431853f146fb274 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431853f1487fd34 INFO:SerialBattery:Testing Renogy
@400000006431853f24c9fddc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431853f24e588a4 INFO:SerialBattery:Testing Ecs
@40000000643185400af9612c INFO:SerialBattery:Testing LltJbd
@40000000643185401b4b925c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000643185401b5cad44 INFO:SerialBattery:Testing Ant
@40000000643185402bb58594 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000643185402bc8e29c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@40000000643185402bd52f0c INFO:SerialBattery:Testing Daly
@40000000643185410158045c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064318541017de3ac INFO:SerialBattery:Testing Daly
@400000006431854111bd86dc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431854111cec4ec INFO:SerialBattery:Testing Jkbms
@400000006431854122395464 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431854122397b74 INFO:SerialBattery:Testing Lifepower
@4000000064318541327f9eb4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064318541329f5bb4 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@400000006431854132b30eac INFO:SerialBattery:Testing Renogy
@400000006431854208b14204 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431854208c92734 INFO:SerialBattery:Testing Renogy
@4000000064318542197b6254 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000643185421995667c INFO:SerialBattery:Testing Ecs
@40000000643185423b47125c INFO:SerialBattery:Testing LltJbd
@4000000064318543101e8c7c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064318543102ef79c INFO:SerialBattery:Testing Ant
@40000000643185432066f1dc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000643185432076341c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000064318543207fec64 INFO:SerialBattery:Testing Daly
@400000006431854331308974 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000643185433142409c INFO:SerialBattery:Testing Daly
@400000006431854405d63594 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006431854405e6d764 INFO:SerialBattery:Testing Jkbms
@4000000064318544160b0b24 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064318544161b81fc INFO:SerialBattery:Testing Lifepower
@4000000064318544263ab47c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064318544264a7b8c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000064318544265456fc INFO:SerialBattery:Testing Renogy
@40000000643185443668cf64 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000643185443678bd84 INFO:SerialBattery:Testing Renogy
@40000000643185450b1570c4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000643185450b2756cc INFO:SerialBattery:Testing Ecs
@40000000643185452c7c67a4 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0

I reinstalled Venus OS from scratch
your program
followed your documentation and Andy Yourtube
But nothin helps

Can you halp
Please
Patrick
But nothing

closed 2023-04-15 by githubfrance

mr-manuel, 2023-04-15:
How did you solve it?

githubfrance, 2023-04-15:
Hi Louis,

I cut the cable from the JKBMS 485 and kept only the side with the connector to the GPS. I connected the ground, TX and TX to the TTL to USB converter.
It has been running flawlessly for two days now.  

[image.png](https://drive.google.com/file/d/13Cw6J_9btVQWCGmTskhvVRrXPdeD2KXe/view?usp=sharing)

I also tested another TTL converter
 
https://drive.google.com/file/d/1Kadc8Wt0xzEfWW1uRZ9p5hc5X2WTZqsp/view?usp=sharing

https://drive.google.com/file/d/1JOqjtiiXrSSm0qrLVHdL2eSq9X5_-2ce/view?usp=sharing

I will now look for a much smaller TTL converter and see if I can welded directly to the raspberry to keep it into the case.

I will see how to start the serial comm port on Venus OS.

Thank you for your work.
Best regards
Patrick

mr-manuel, 2023-04-15:
Ok, so related to https://github.com/Louisvdw/dbus-serialbattery/discussions/524. Thanks

hahnkarl, 2023-10-29:
I dont know if I'm on the right topic but i have issue ( connectet via BLUETOOTH) with 2 of 3 JK devices. one is working well 2 are working also but i only get the voltage of each cell but not the "rest" of datas. the working jk is SW 10.09, the not working SW11.25 maybe you can help me , thanks