Louisvdw/dbus-serialbattery #522: MAX_BATTERY_CHARGE_CURRENT should be retrieved from the system settings

jdeus, 2023-04-10:
**Is your feature request related to a problem? Please describe.**
MAX_BATTERY_CHARGE_CURRENT is currently a hard setting. However it's meaning should be the same as the mac charge current defined in VenusOS, and thus its value should be retrieved from the VenusOS's config.

**Describe the solution you'd like**
See above

**Describe alternatives you've considered**
The setting could eventually be retrieved from the DVCC's settings. But not everyone has DVCC enabled.

mr-manuel, 2023-04-10:
What for those who have multiple batteries in parallel? How you would handle it, if one battery has a max charge current limit of 40 and the other 60?

What is your application/system to understand your request better?

mr-manuel, 2023-04-10:
Normally you don‘t set this values. You only enable DVCC that the driver can work with this values.

Many are using Node-RED and [venus-os_dbus-mqtt-battery](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery) to aggregate batteries.

jdeus, 2023-04-10:
So what's the use case of such config variables exactly ?

mr-manuel, 2023-04-10:
See [8. DVCC - Distributed Voltage and Current Control](https://www.victronenergy.com/media/pg/Venus_GX/en/dvcc---distributed-voltage-and-current-control.html)

closed 2023-04-14 by mr-manuel