Louisvdw/dbus-serialbattery #526: Low Battery Voltage Alarm

kakariki1, 2023-04-13:
Hello,

I've always got "Low Battery Voltage Alarms" allthough the battery was not empty for sure.
There are two more posts regardings this topic but no solution: https://github.com/Louisvdw/dbus-serialbattery/issues/407, https://github.com/Louisvdw/dbus-serialbattery/issues/297

Now I found the root cause for this issue: This happens only if MaxDischargeCurrent is zero (or below 0.1A).

My solution is to limit MaxDischargeCurrent to >= 0.1 and additionally enable "Charge voltage control management" by setting "CVCM_ENABLE = True" in utils.py
This works flawlessly for me for several weeks now.
Because I'm running different systems in different configurations I have not done any changes to dbus-serialbattery directly, but at an higher-order level in dbus-aggregate-batteries instead (see https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/31).

Maybe it helps somebody.

Best regards,
lakeroe

mr-manuel, 2023-04-13:
Maybe you know why your max discharge current went to 0? Was it because of cell voltage, temperature or something else?

kakariki1, 2023-04-13:
It was definitly because of cell voltage ...

larsenf, 2023-04-13:
Im having the same issue but my max discharge setting is 120 because im using on offgrid system. It triggers at i think 100% SOC and whenever i stop charging. The alarm comes from the multi and stops my ac output, it doesnt happen when the venus device is off. CCCM_CV_ENABLE has been false but i changed it to true and will see if i still get the alarm.

Dr-Gigavolt, 2023-04-13:
I had this alarm issue too. I thought it is comming directly from Multis and had no explanation for it. Since I disabled everything in Serial Battery (I calculate parameters with my Aggregate Batteries anyway) it disapeared. But I updated Venus at the same time and therefore had not only single change.

mr-manuel, 2023-04-14:
@larsenf it would also be interesting, if the larm also shows, if you change 0 with 0.1