Louisvdw/dbus-serialbattery #528: driver without ESS control

Micha854, 2023-04-13:
i have a pylontech battery and a DIY with daly bms. Both are connected to the raspberry via CAN.

I combine both with the battery aggregator. works fine

I would now like to use your seral driver only for the display (evaluation).

currently the system goes completely crazy when I activate the serial driver!

is there a possibility that the driver does nothing but deliver data?

mr-manuel, 2023-04-13:
Please see https://github.com/Louisvdw/dbus-serialbattery/discussions/517.

At the moment this is not possible.

mr-manuel, 2023-04-13:
@Micha854 would you see the data of the battery in the Venus OS GUI or you need the data only over MQTT? I don‘t think that the first is possible, since this is not the way Venus OS works.

You could try to play with the dbus values
```
/State
/SystemSwitch
```
but it would be a dirty workaround.

Micha854, 2023-04-13:
I found the following and it is reproducible:

1. If you have activated the driver, restart the GX device.
2. wait until the bms is displayed in venus
3. deactivate three!
4. now the bms stays there until the next restart of the GX device and you have all the data

Can't you build something there?

mr-manuel, 2023-04-13:
Deactivate what exactly where and how?

mr-manuel, 2023-04-13:
> Deactivate what exactly where and how?

So what is now clear, but where and how not. Please clarify. There are a lot of possibilities to disable the driver.

Additionally which driver version and Venus OS are you using?

mr-manuel, 2023-04-13:
As long as the data is in the dbus Venus OS will use it.

have you tried to select the battery monitor manually in Venus OS GUI?

Micha854, 2023-04-13:
I have the latest version of the serial driver. I tried both venus os 2.93 and the new beta.

but as soon as i deactivated the driver, the system ran normally again.

with the driver activated it was chaos

Micha854, 2023-04-14:
UPDATE:

I got the system up and running now!
in the utils.py file I have adjusted the following values:

CCCM_CV_ENABLE = False
DCCM_CV_ENABLE = False
CCCM_T_ENABLE = False
DCCM_T_ENABLE = False
CCCM_SOC_ENABLE = False
DCCM_SOC_ENABLE = False
CVCM_ENABLE = False
INVERT_CURRENT_MEASUREMENT = -1

Here is my config again:

- Venus OS 3.00 beta 32
- pylontech battery connect via waveshare can-hat
- DIY battery with daly bms connected via can from daly interface board to canable pro usb adapter and from interface board connect the rs485 to usb for serial driver
- BatteryAggregator from @pulquero summarizes both batteries (batteries via USB are excluded in the config)

in DVCC "Controlling BMS" I have set "BatteryAggregator".
This is also my battery monitor

mr-manuel, 2023-04-14:
So this is solved for you?

Micha854, 2023-04-14:
Yes, I have now found the right solution for me

Has been running very stable and flawless since yesterday

Thanks!

closed 2023-04-14 by mr-manuel

mr-manuel, 2023-04-15:
I have a update for you.

The battery monitor is not the same as the BMS selected.

With Venus OS >3.00 you can select the BMS that should be used for DVCC:
[image]

Source: https://community.victronenergy.com/questions/199460/multiple-battery-monitor-available-only-selected-a.html?childToView=199503#comment-199503

Micha854, 2023-04-15:
@mr-manuel yes, of couse. I also wrote that above. I use the "BatteryAggregator" as BMS

A combination of Daly + Pylontech

All parameters are the same and then summarized!