Louisvdw/dbus-serialbattery #544: Write SOC to bms?

thomasinaz41213, 2023-04-21:
**Is your feature request related to a problem? Please describe.**
Some BMS's are very inaccurate when it comes to SOC. For example I am using a 500amp Daly which has been installed for about 3 weeks now. The SOC as reported by the Daly is about 15% lower than the actual SOC as tracked by the smart shunt which is pretty accurate. This equates to about 5% per week and I have been plugged into shore power for most of that. The problem with the discrepancy may actually get worse as I spend more time off grid this summer.

**Describe the solution you'd like**
Is there any way possible, through the use of this driver, to write the SOC value to the BMS? It can be done through the same uart connection using PC software so the write capability is definitely there. I could disconnect the uart adapter periodically and use the PC software but that's a hassle. If there's any way to facilitate this through the driver it could then possibly be automated through the use of node red, essentially creating a sync mechanism between the smart shunt and the BMS.

mr-manuel, 2023-04-21:
Hi,

currently the driver can only read data from the BMS. It‘s not planned in near future that also writing is possible.

To solve your issue you can use a virtual battery and feed it with combined data over MQTT from Node-RED. Here a is link to the repository https://github.com/mr-manuel/venus-os_dbus-mqtt-battery.

thomasinaz41213, 2023-04-21:
Currently I'm using the smart shunt as battery monitor in Venus and using the Daly driver for dvcc ESS control. It works fine as is but just annoying that the Daly SOC is nowhere near correct. Call me OCD.

closed 2023-04-22 by mr-manuel