Louisvdw/dbus-serialbattery #547: Instance number changes at reboot

thomasinaz41213, 2023-04-24:
My apologies if this has been covered elsewhere. I searched but didn't find anything.
**Describe the bug**
If the cerbo is reboot for any reason, often times the bms gets assigned a different instance number which then requires changing settings and reconnecting nodes in node red.

**To Reproduce**
Steps to reproduce the behavior:
Reboot cerbo, then battery alarm 'low voltage' because bms has been assigned a different number. Seems to bounce back and forth between 1 and 2.

**Expected behavior**
I would expect the bms to always be assigned the same number.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**VenusOS (please complete the following information):**
 - Device type: cerbo gx
 - Firmware Version: same behavior on 2.93 and beta 3.00~32

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: Daly
 - Cells: 8
 - Interface: Daly uart to usb

**Additional context**
I have a rs485 to USB adapter plugged into the first USB port which runs a external relay board. Plugged into the second USB port is a powered USB hub with a mppt on ve.direct to USB, a USb bt dongle, tank140, and the uart Daly adapter. All other devices survive reboots with no issues.

I have posted this issue on the victron board and victrons response is that it is a bms driver issue.

ramack, 2023-04-24:
I face similar problems and think that we should allow a dedicated mapping from serialnumbers as read by the low level drivers to instance numbers. That would also help in setups with multiple batteries.

mr-manuel, 2023-04-24:
Hello,

please see https://github.com/Louisvdw/dbus-serialbattery/discussions/481 and https://github.com/Louisvdw/dbus-serialbattery/issues/100.

closed 2023-04-24 by mr-manuel