Louisvdw/dbus-serialbattery #576: Uninstalling driver bricked my cerbo

thomasinaz41213, 2023-04-26:
Cerbo gx firmware 3.00~32 tried to uninstall the driver following the directions on the wiki. I copy and pasted and got no errors in the console so I'm sure I didn't enter anything wrong. After following all the instructions, several items on the gui were red. I reboot the cerbo hoping it would straighten itself out but it never came back on. Just a flashing red led and no display, no more console, no wifi or Bluetooth. Bricked unit. Great, now I get to go through warranty.

mr-manuel, 2023-04-26:
Hello,

did you run this?

```bash
DRIVERNAME=dbus-serialbattery

# handle read only mounts
sh /opt/victronenergy/swupdate-scripts/remount-rw.sh

# remove files
rm -rf /data/conf/serial-starter.d
rm -rf /opt/victronenergy/service/$DRIVERNAME
rm -rf /opt/victronenergy/service-templates/$DRIVERNAME
rm -rf /opt/victronenergy/$DRIVERNAME

# kill if running
pkill -f "python .*/$DRIVERNAME.py"

# remove install-script from rc.local
sed -i "/sh \/data\/etc\/$DRIVERNAME\/reinstalllocal.sh/d" /data/rc.local
```

mr-manuel, 2023-04-26:
Probably the line `DRIVERNAME=dbus-serialbattery` was not executed and now the folder `/opt/victronenergy/` was deleted.

Did you try this? https://community.victronenergy.com/questions/66811/cerbo-not-booting-flashing-orange-light.html

mr-manuel, 2023-04-26:
Can you still connect over SSH? When this happened to me, this was still possible.

Upload this file via SFTP to `/tmp` and rename it to `check-updates.sh`:

[check-updates.sh.txt](https://github.com/Louisvdw/dbus-serialbattery/files/11329262/check-updates.sh.txt)

Connect over SSH and run `bash /tmp/check-updates.sh -update -force`

thomasinaz41213, 2023-04-26:
Yes those are the commands that I ran. I did come across those threads in my frantic search for a fix. The only serial adapter I have is a rs485 2 wire (rx/tx only). I'm sure I could figure out how to get the serial connection working but beyond that I would be lost without explicit step by step instructions. I'm not really a Linux guy. I know just enough to get myself into trouble, as evidenced by my previous post, lol.

thomasinaz41213, 2023-04-26:
> Can you still connect over SSH? When this happened to me, this was still possible.
> 
> Execute this to force a firmware update and repair:
> 
> ```shell
> wget -O /tmp/check-updates.sh https://raw.githubusercontent.com/mr-manuel/venus-os_helpful-scripts/master/swupdate-scripts/check-updates.sh && bash /tmp/check-updates.sh -update -force
> ```

I no longer have access to ssh. I'm assuming running the update through the serial console is not an option since the os does not boot?

mr-manuel, 2023-04-26:
If the OS does not boot, than you have to change the boot partition. When you update Venus OS at least once, then the updater creates a second partition as "backup".

thomasinaz41213, 2023-04-26:
> If the OS does not boot, than you have to change the boot partition. When you update Venus OS at least once, then the updater creates a second partition as "backup".

Ok 2 questions then...
1. Can I connect to the serial connection with the 2 wire rs485 adapter and if so, what do I use to access the terminal from within windows?
2. How do I change the boot partition?

It would be great if Victron could use the onboard button on the cerbo to initiate a boot partition swap for example hold the button while inserting the power connection.

mr-manuel, 2023-04-26:
1. No, you need a USB to serial (RS232) adapter.
2. Did you try to see, if the device is getting an IP address from DHCP or if you can ping the static IP, if you gave one?
3. Unfortunately I don't know it yet. I'm trying with my Raspberry to achieve this. Maybe someone other can help with that?

One firmware is on `mmcblk0p2` and the other on `mmcblk0p3`, but I did not found, where it tells which partition is mounted.

## firmware 1
```bash
root@raspberrypi:~# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 14.9G  0 disk
├─mmcblk0p1 179:1    0 41.3M  0 part /u-boot
├─mmcblk0p2 179:2    0  4.9G  0 part /
├─mmcblk0p3 179:3    0  4.9G  0 part
└─mmcblk0p4 179:4    0  5.1G  0 part /data

root@raspberrypi:~# cat /etc/fstab
/dev/root / auto defaults 1 0
proc                 /proc                proc       defaults              0  0
devpts               /dev/pts             devpts     mode=0620,gid=5       0  0
tmpfs                /run                 tmpfs      mode=0755,nodev,nosuid,strictatime 0  0
tmpfs                /var/volatile        tmpfs      defaults              0  0
/dev/mmcblk0p1       /u-boot              vfat       defaults              0  0
/dev/mmcblk0p4       /data                ext4       noatime               0  2
```

## firmware 2
```bash
root@raspberrypi:~# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 14.9G  0 disk
├─mmcblk0p1 179:1    0 41.3M  0 part /u-boot
├─mmcblk0p2 179:2    0  4.9G  0 part
├─mmcblk0p3 179:3    0  4.9G  0 part /
└─mmcblk0p4 179:4    0  5.1G  0 part /data

root@raspberrypi:~# cat /etc/fstab
/dev/root / auto defaults 1 0
proc                 /proc                proc       defaults              0  0
devpts               /dev/pts             devpts     mode=0620,gid=5       0  0
tmpfs                /run                 tmpfs      mode=0755,nodev,nosuid,strictatime 0  0
tmpfs                /var/volatile        tmpfs      defaults              0  0
/dev/mmcblk0p1       /u-boot              vfat       defaults              0  0
/dev/mmcblk0p4       /data                ext4       noatime               0  2
```

In the meanwhile you could try to establish a serial connection. If you have Windows you can use [putty ](https://the.earth.li/~sgtatham/putty/latest/w64/putty.exe) for that.

> It would be great if Victron could use the onboard button on the cerbo to initiate a boot partition swap for example hold the button while inserting the power connection.

Feel free to open a question in the Victron Community for that. Indeed that would be a great idea.

thomasinaz41213, 2023-04-26:
I did open a question on the victron forum for that. We'll see if there is any response. In the meantime, I have ordered a replacement cerbo which should be here day after tomorrow. My distributor has submitted for warranty replacement. I'm a little hesitant to try the driver again for fear of a repeat occurrence. All I was trying to do was get a fresh install of the latest release. The last few days I have been getting disconnects from the Daly after about 24hrs. Worked fine for a few weeks without a single coms failure then all of a sudden it started.

mr-manuel, 2023-04-26:
I already changed the uninstall script, so you cannot make any mistake again.

If you altrough are able to delete more than necessary and see several items on the GUI red, then DO NOT reboot the device. If the device is still online you can force a reinstall of the firmware. See https://github.com/Louisvdw/dbus-serialbattery/issues/576#issuecomment-1522807185

Adittionally to have a clean install you also can force a reinstall of the firmware with `/opt/victronenergy/swupdate-scripts/check-updates.sh -update -force`, but before delete the `/data/rc.local` file.

Nice to hear, that you got a new device under warranty!

thomasinaz41213, 2023-04-26:
I just ordered a replacement and will have to wait and see if I get the bricked unit replaced under warranty. ESS is down along with many other things that now rely on the cerbo for control. At least I am able to put all the blue stuff in stand alone so I'm not totally dead in the water.

thomasinaz41213, 2023-05-01:
I have already swapped the device but I have ordered a serial adapter and I am going to try to get serial console access and see if I can recover the device.

thomasinaz41213, 2023-05-01:
I'm still waiting for the adapter but I will report back once I have it and have a chance to check it out.

thomasinaz41213, 2023-05-04:
Got the serial adapter today and was able to connect to the console. As stated in a previous post opt/victronenergy had been deleted so there was no way to run any of the update scripts. I was able to copy the scripts from a working system and chmod to set file permissions then I was able to switch to the other partition and successfully boot the cerbo. It is alive again.

mr-manuel, 2023-05-04:
Could you describe how you did it? Which adapter did you ordered exactly and how did you copy the files? This will help also others with the same problem.

thomasinaz41213, 2023-05-04:
This link explains how to connect to the serial console https://www.victronenergy.com/live/ccgx:root_access#:~:text=CCGX%20Serial%20Console.-,Serial%20Console%20on%20Cerbo%20GX,are%20pins%204%20and%205.
I used the adapter linked in the instructions

I used putty to connect to the console. Once connected I copied opt/victronenergy/swupdate-scripts (this is where all the update scripts are located) and opt/victronenergy/version onto the cerbo, then chmod a+x to all the files.

Then enter the following in console
/opt/victronenergy/swupdate-scripts/set-version.sh 2
Once the cerbo reboots do not swap back to the other image, force an update so a new image is written to the other partition. I just did an online update from settings.

mr-manuel, 2023-05-04:
Perfect, thanks!

closed 2023-05-04 by mr-manuel