Louisvdw/dbus-serialbattery #583: No response on USB/SD installation

josmeijer, 2023-04-26:
Hi,

I tried to install either by USB,SD card, and SSH but no success to report

1. I inserted the USB stick with the driver  (venus-data.tar.gz) and the DALY RS485 cable (other end in the bms) USB plug both in the back of the ColorControl
2. I went to the settings>general menu on the GX and did a reboot
3. I waited patiently for the reboot, but in the devices menu no Serial Battery appeared.
4. I did a second reboot, as the driver might need that after copyying to main memory.
5. I got no visible proof of the existence of the driver, the device list didn't show it.
6. I did some tampering with setting battery monitor from auto to off, but no result.
7. I repeated the procedure with an SD card, I formatted it (FAT32) copied and carefully checked the files on it, even unmounted it properly (can't remember doing that for ages), inserted it in the CCGX. Same result.

**Screenshots**
[image]
this is the SD card after the operation. Apparently it was in contact with the CCGX, which wrote a backupfile on it.

VenusOS 
 - Device type: ColorControl GX
 - Firmware Version 3.0

 - BMS/Battery type: DALY 250A smart BMS
 - Cells: 16 LiFePo 230 AH
 - Interface:  USB-RS485.

mr-manuel, 2023-04-26:
Hello, please check the [How to troubleshoot](https://louisvdw.github.io/dbus-serialbattery/troubleshoot/) page.

The driver probably is already installed, but did not find any BMS and therefore is not showing up in the GUI.

mr-manuel, 2023-04-26:
1. What adapter are you using?
2. Is there a TX/RX light?

josmeijer, 2023-04-26:
I am using the RS485-USB adapter that was supplied by DALY. There is no
indication leds on this one.
But then, I realised that it is the same cable that I connect to my
notebook

josmeijer, 2023-04-27:
Apparently the driver did not install from USB. I did succeed installing it via SSH (after some struggling with the password whichshould be 6 characters minimal) 
I do have serialbattery running now, but some parameters that do show in de DALY PC software are not displayed in the GX panel.
[image]

mr-manuel, 2023-04-28:
Please install the [nightly version](https://louisvdw.github.io/dbus-serialbattery/general/install#install-over-ssh-nightly-versionbeta-test). It contains a fix for that.

closed 2023-05-01 by mr-manuel