Louisvdw/dbus-serialbattery #589: Current dumping to grid with no solar input after D-bus-serialbattery installation

josmeijer, 2023-04-27:
**Describe the bug**
Afte finally (after some bumps) finishing the installation of Dbus SB I found that a sudden Grid feed-in of >4000w was taking place.

I just installed dbus-serial battery drivers and did not change any other parameters.
I allways have AC and DC grid feed in switched on. I was forced to switch them off to avoid a total drain of the battery (I presume)
There are no alarms or messages.
[image]
[image]

**VenusOS 3.0
 - Device type: Color Control GX
 - Firmware Version 3.0~36

**Battery/BMS 
 - Daly 250A smart BMS
 - Cells: 16 LiFePo 250 Ah
 - Interface: USB-RS485 (factory standard DALY)

mr-manuel, 2023-04-28:
Which devices do you have?

josmeijer, 2023-04-28:
EasySolar 48/5000/70

josmeijer, 2023-04-28:
further I have this funny inverted display of discharging since the sun is up and charging when I make my coffee with 3kw draw.

[image]
[image]

mr-manuel, 2023-04-28:
> further I have this funny inverted display of discharging since the sun is up and charging when I make my coffee with 3kw draw.

Please check the FAQ for this.

josmeijer, 2023-04-28:
While sitting with the coffee I just mentioned I did some thinking:
can it be, that the inverted charge/discharge will as well be the cause of the grid-dumping? 
I don´t know how the algorithm works, but I can imagine this. And if so, you get an avalanche-effect as the more you dump, the more you "charge" .

mr-manuel, 2023-04-28:
Yes that could be.

[Why is the battery current inverted?](https://louisvdw.github.io/dbus-serialbattery/troubleshoot/faq#why-is-the-battery-current-inverted)

josmeijer, 2023-04-28:
so "Some Daly BMS send the current as inverted value. This can be correted by setting INVERT_CURRENT_MEASUREMENT to -1"
That sounds easy, as long as you know where to find (-and change) this parameter.
And, I only have SSH acces, as the USB/SD install failed, so changing it in the tar.gz file is no real option.
I do have some experience but that was in the 80's of last century, working CP/M on a Z80 processor. So, though I surprised myself by memorizing quite some of the commands, my knowledge is yet a little rusty, so some direction would be appreciated

mr-manuel, 2023-04-28:
Take a look at [How to change the default limits](https://louisvdw.github.io/dbus-serialbattery/general/install#how-to-change-the-default-limits) and [How to edit utils.py or config.ini](https://louisvdw.github.io/dbus-serialbattery/general/install#how-to-edit-utilspy-or-configini).

josmeijer, 2023-04-29:
typical RTFM :-)  
It was all there
And great that all my 80's 
[image]
wordstar commands still work in Nano

josmeijer, 2023-04-29:
[image]
typical RTFM case :-)  It was all there.
And great that all my 80's wordstar commands still work in nano.
thnx for guiding me.

mr-manuel, 2023-04-29:
You’re welcome. I modified also the docs to make them more clear and easy to understand ;-) great that it works now! All problems solved?

josmeijer, 2023-04-29:
It seems so. 
I was still thinking: can it be that the "some daly BMS's invert charge/discharge" is related to the settings of the inverter protocol (bottom right on the picture)? 

Mine is still set on Victron from an attempt to realise CAN bus communication 

[image]

mr-manuel, 2023-05-01:
Please open a discussion for that.

closed 2023-05-01 by mr-manuel