Louisvdw/dbus-serialbattery #590: MNB BMS still supported?

mr-manuel, 2023-04-28:
Is the MNB BMS still supported? It was first disabled and then removed.

* disable MNB battery by default https://github.com/Louisvdw/dbus-serialbattery/commit/65241cbff36feb861ff43dbbcfb2b495f14a01ce

* remove duplicate MNB lines (there were no duplicates?) https://github.com/Louisvdw/dbus-serialbattery/commit/23afec33c2fd87fd4d4c53516f0a25f290643c82

Is someone using this BMS with this driver?

Louisvdw, 2023-05-02:
This is still valid. The user needs to install the SPI and another reference that is required before it can be used. So by default it will fail if activated and that is why it is disabled. I'll add a link later to the steps (need to go and search for that post)

mr-manuel, 2023-05-02:
Ok. Could you add what a user have to do to the [BMS specific install settings](https://louisvdw.github.io/dbus-serialbattery/general/install#bms-specific-settings)?

mr-manuel, 2023-05-28:
I found only this https://github.com/Louisvdw/dbus-serialbattery/pull/27

MikeD2-mnb, 2023-09-17:
As this code is specific to one series of chips, at the moment I'm willing to support this on a case by case basis. I do have hardware : Built & tested boards available, or I can make the pcb available for on line purchase, and you can build your own. Code is also subject to revision for python3 language support in Venus os. Note that there are some python libraries that need to be installed on the os manually: the code does NOT do this for you.

mr-manuel, 2023-09-18:
Since it‘s currently nearly impossible, I will remove it for this time. See [here](https://community.victronenergy.com/comments/231924/view.html).

closed 2023-09-18 by mr-manuel