Louisvdw/dbus-serialbattery #596: installed capacity displays 50Ah

josmeijer, 2023-04-30:
[image]
 
bug is not the right word, I would say inconvenience or "something that puzzles me"
I ran through all possible parameters (I could find) but I see no way to tell the d-bus serialbattery driver that I have a 230Ah battery bank installed.
The available capacity does match reality though.

mr-manuel, 2023-04-30:
Did you saw the template for a new issue? If not please fill it out and explain shortly how you opened the issue, so we can improve the workflow. If yes there is a reason why we need this data to help you :-)

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Venus OS (please complete the following information):**
 - Device type: [e.g. VenusGX, Cerbo, Raspberry Pi]
 - Firmware Version [e.g. 262]

**dbus-serialbattery**
 - Version: [e.g. v0.14.3]

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: [e.g. LTT, Daly]
 - Cells: [e.g. 16]
 - Interface: [e.g. USB-RS485, USB-RS232, Ve.Direct]

**Additional context**
Add any other context about the problem here.

towatai, 2023-05-09:
You have to Change the Capacity-Size in the config.default.ini from 50 to 230.

; -- Daly settings
; Battery capacity (amps) if the BMS does not support reading it
BATTERY_CAPACITY = 230

mr-manuel, 2023-05-09:
If it's a Daly, then this is solved. With the published beta, also the capacity from the Daly are read from the BMS.

closed 2023-05-09 by mr-manuel

josmeijer, 2023-06-17:
a little late, but does this imply I should reinstall the latest version of the driver (or manually change the value)?

mr-manuel, 2023-06-18:
In the meanwhile a lot changed. I would recommend you to use latest Venus OS with latest driver version.

josmeijer, 2023-06-18:
I have auto-update on in the CCGX anyway. I'll give the "install from SD" another chance, hopefully that functioning is one of the many changes :-)

mr-manuel, 2023-06-18:
Why aren't you updating over SSH?

josmeijer, 2023-06-18:
I was curious whether the new firmware would have solved the ability to
install from SD card (it didn't work on the 0.nn version).
And it is very easy on the CCGX as there is a micro SD slot on the front.
I did all the other adjustments via SSH though.