Louisvdw/dbus-serialbattery #610: jkbms working with CAN interface

IrisCrimson, 2023-05-05:
### Is your feature request related to a problem? Please describe.

Hi i have a running jkbms CAN version now running since more than 10 month.
If I get write access to the repo I could provide my code.

mr-manuel, 2023-05-05:
You don't need to have write access to do that. Please fork the repository, add your driver to the `jkbms_ble` branch and then open a PR.

closed 2023-05-05 by mr-manuel

seamaster101, 2023-05-09:
@IrisCrimson 
what kind of JKBMS do you have? 
Do you know if CAN is an option for the JK_B2A8S20P BMS?
Does anyone has interfaced that one?

pos-ei-don, 2023-05-09:
@IrisCrimson, have you connected more than one bms with can? How do you Set the can bus ID?
Very interesting!

mr-manuel, 2023-05-12:
> @IrisCrimson what kind of JKBMS do you have? Do you know if CAN is an option for the JK_B2A8S20P BMS? Does anyone has interfaced that one?

It's not integrated by default. You have to select the can option before buying. The script has to be updated to the latest version.

https://github.com/IrisCrimson/dbus-serialbattery/blob/30fb705d801ea88deac762c354457b130ca89f3f/etc/dbus-serialbattery/jkbms_can.py

mr-manuel, 2023-09-17:
Please try the latest `nightly` from the `dev` branch. To enable CAN add `CAN_PORT = can0` to the `config.ini`.

DirkReudenbach, 2023-09-20:
Thats interesting new feature. 
Actually I'm running as serial driver via RS485 but I'd like to test it because of the trouble with USB Ports, My JKBMSes 4x B2A24S15P have RS485 and should have Canbus as well
How do I connect this best to my Cerbo?

pos-ei-don, 2023-09-26:
And how do you Set the canbus id?

mr-manuel, 2023-09-26:
I have no idea, hopefully @IrisCrimson can answer that.

IrisCrimson, 2023-11-19:
Hi
regarding the CAN ID, as far as I know there is no way to change this.
My solution was to build a small CAN gateway. Based on an old PRI version
1.2 B and several MCP2515 shields.
I will push the script and the howto to git within the next days.....

Meanwhile i have as well tested the jkbms_can version Manuel merged to the
dev branch.
All my findings/fixes and some extensions that are needed to use more than
one CAN bms at different CAN ID's could be found at following GIT Branch:
https://github.com/IrisCrimson/dbus-serialbattery/tree/jkbms_can_2
Within this branch you could find as well the description how to build the
needed can cable:
https://github.com/IrisCrimson/dbus-serialbattery/blob/jkbms_can_2/docs/how_to_cable_jkbms_can_to_victron_can.txt

On top of that i created an master/aggregation BMS this I have pushed to
following branch:
https://github.com/IrisCrimson/dbus-serialbattery/tree/dbus-masterbattery
-> Maybe this could fully merged into the dbus-serialbattery.

[image: grafik.png]

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virenfrei.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#m_3103273102869763835_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>