Louisvdw/dbus-serialbattery #612: Device new name does not survive reboot

seamaster101, 2023-05-06:
### Describe the bug

I have two JKBMSs connected and try to rename then Battery1 and Battery2 respectively

The names are excepted and presented properly till the unit is rebooted. 
[image]

Once the unit is rebooted it goes back to default SerialBattery(Jkbms) name
 
[image]

### How to reproduce

rename SerialBattery
reboot device

### Expected behavior

Needs to keep the new devices name after the reboot

### Driver version

0.14.3

### Venus OS device type

Please select

### Venus OS version

V3.0

### BMS type

JKBMS / Heltec BMS

### Cell count

4

### Connection type

Serial USB adapter to RS485

### Config file

```ini
not modified
```

### Relevant log output

```shell
not sure what is needed here
```

### Any other information that may be helpful

can provide if else is needed

pos-ei-don, 2023-05-06:
Same here in this Release. As a workaround you can Set the Name within NodeRed and Mqtt.

seamaster101, 2023-05-06:
@pos-ei-don I'm glad that you managed to confirm the renaming issue. 
Unfortunately I don't use nodeRed or MQTT

mr-manuel, 2023-05-06:
Duplicate of https://github.com/Louisvdw/dbus-serialbattery/issues/100

closed 2023-05-06 by mr-manuel