Louisvdw/dbus-serialbattery #613: JKBMS SerailBattery many "Unknown" in parameters

seamaster101, 2023-05-06:
### Describe the problem

When reviewing the connected JKBMS via SerialBattery 
i noticed the following fields reported as unknown:
1. Switch - Unknown
2. State - Unknown 
3. Relay state - Unknown
4. Alarm State - Unknown
Under Diagnostics (it is listed actually twice with different sub-menus)
     Alarm -Unknown
     Main Switch - Unknown
     Heater  - Unknown
     Internal Fan - Unknown 
I understand that some of this is because data is not provided by the BMS but i care the most about the Alarm State
[image]

I presume the alarm state should be known as under Alarms we have OK for everything, 
[image]

so AlarmState should be "OK"
[image]

### Driver version

0.14.3

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.0

### BMS type

JKBMS / Heltec BMS

### Cell count

4Cell multiple batteries

### Connection type

Serial USB adapter to RS485

### Config file

```ini
original
```

### Relevant log output

```shell
can provide what is needed
```

mr-manuel, 2023-05-06:
Thats already fixed in the next version. You can try the nightly to solve hat.

closed 2023-05-06 by mr-manuel

seamaster101, 2023-05-08:
@mr-manuel i did update to this version:
[image]
still looses the name after reboot

mr-manuel, 2023-05-08:
See https://github.com/Louisvdw/dbus-serialbattery/issues/100