Louisvdw/dbus-serialbattery #619: SSH Update does not work

UdoD5, 2023-05-06:
### Describe the problem

See the attached log screenshot:

[image]

### Driver version

0.14.3

### Venus OS device type

Cerbo GX

### Venus OS version

v2.94

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
# -*- coding: utf-8 -*-
import logging
import serial
from time import sleep
from struct import *
import bisect

# Logging
logging.basicConfig()
logger = logging.getLogger("SerialBattery")
logger.setLevel(logging.INFO)

# battery types
# if not specified: baud = 9600
battery_types = [
    {'bms' : "LltJbd"},
    {'bms' : "Ant", "baud" : 19200},
    {"bms" : "Daly", "address" : b"\x40"},
    {"bms" : "Daly", "address" : b"\x80"},
    {"bms" : "Jkbms", "baud" : 115200},
#    {"bms" : "Sinowealth"},
    {"bms" : "Lifepower"},
    {"bms" : "Renogy", "address": b"\x30"},
    {"bms" : "Renogy", "address": b"\xF7"},
    {"bms" : "Ecs", "baud" : 19200},
#    {"bms" : "MNB"},
]

# Constants - Need to dynamically get them in future
DRIVER_VERSION = 0.14
DRIVER_SUBVERSION = '.3' 
zero_char = chr(48)
degree_sign = u'\N{DEGREE SIGN}'

# Choose the mode for voltage / current limitations (True / False)
# False is a Step mode. This is the default with limitations on hard boundary steps
# True "Linear"    # New linear limitations by WaldemarFech for smoother values
[... 245 lines cut ...]

        count = 0
        data = bytearray(res)
        while len(data) <= length + length_check:
            res = ser.read(length + length_check)
            data.extend(res)
            #logger.info('serial data length ' + str(len(data)))
            sleep(0.005)
            count += 1
            if count > 150:
                logger.error(">>> ERROR: No reply - returning [len:" + str(len(data)) + "/" + str(length + length_check) + "]")
                return False

        return data

    except serial.SerialException as e:
        logger.error(e)
        return False
```

### Relevant log output

```shell
Where do I find the log output?
```

mr-manuel, 2023-05-06:
Indeed there is a small issue. We will fix this.

mr-manuel, 2023-05-06:
In the meanwhile you can run this:

```bash
wget -O /tmp/install.sh https://raw.githubusercontent.com/mr-manuel/dbus-serialbattery/master/etc/dbus-serialbattery/install.sh
bash /tmp/install.sh
```

Select `4` and paste the URL of the `venus-data.tar.gz ` you want to install.

peregrines2, 2023-05-07:
> In the meanwhile you can run this:
> 
> ```shell
> wget -O /tmp/install.sh https://raw.githubusercontent.com/mr-manuel/dbus-serialbattery/master/etc/dbus-serialbattery/install.sh
> bash /tmp/install.sh
> ```
> 
> Select `4` and paste the URL of the `venus-data.tar.gz ` you want to install.

May I ask how the URL would look like to install the new v1.0.0beta20230505 (https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v1.0.0beta20230505)
I seems like cant figure it out  because I keep getting an error :( 
Thank You!!

jeroenkl, 2023-05-07:
> > In the meanwhile you can run this:
> > ```shell
> > wget -O /tmp/install.sh https://raw.githubusercontent.com/mr-manuel/dbus-serialbattery/master/etc/dbus-serialbattery/install.sh
> > bash /tmp/install.sh
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Select `4` and paste the URL of the `venus-data.tar.gz ` you want to install.
> 
> May I ask how the URL would look like to install the new v1.0.0beta20230505 (https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v1.0.0beta20230505) It seems like cant figure it out because I keep getting an error :( Thank You!!

https://github.com/Louisvdw/dbus-serialbattery/releases/download/v1.0.0beta20230505a/venus-data.tar.gz

mr-manuel, 2023-05-07:
Go to [Releases](https://github.com/Louisvdw/dbus-serialbattery/releases) -> choose the release you are interested in and click on `Assets` -> Right click on `venus-data.tar.gz ` and copy the URL.

jandbjandb, 2023-05-07:
I have very little knowledge of all what you're discussion here about, but i like the Serial-battery to work on my JBD BMS and Cerbo-gx. Please let us know when the links work again, so I can simply copy them into putty and my Cerbo
I have no idea what to do if the original straight forward explanation from Louis VDW doesn't work, ... sorry for that...

much appreciated !!!

mr-manuel, 2023-05-07:
I opened already a PR: https://github.com/Louisvdw/dbus-serialbattery/pull/620

closed 2023-05-08 by mr-manuel

mr-manuel, 2023-05-08:
It's fixed

jandbjandb, 2023-05-09:
unfortunately, i still get an error message...

[image]

jandbjandb, 2023-05-09:
I needed a few tries, as i got some error messages on Putty, but finally after more than a few trials, i managed to get true and now i have my JBD BMS in the system integrated. Hopefully i manage to set all other parameters correctly .
Many thanks to the developers and software geniuses who made this possible !

[image]

mr-manuel, 2023-05-09:
Did you install the beta or `v0.14.3`?

mr-manuel, 2023-05-09:
> when you say "the driver first" do you mean i need to update (if any) the dbus driver before the new Venus OS ?

Yes. Else you will have problems with Venus OS 3.00