Louisvdw/dbus-serialbattery #624: Installation manuals

Jaco-TTT, 2023-05-08:
### Describe the bug

After a few years, I have re-installed the driver again. Ran without it for years, and now need to view the voltages of 18 cells on the Cerbo remotely.

**The latest manual:**
Install does not work out of the box. Had to trace the commands and what to do when.
Instructions to check for problems need to be updated in the manuals.

**Also the directories. For example:**
All old and some new files under:  /data/etc/dbus-serialbattery
All the newly installed files:  /opt/victronenergy/dbus-serialbattery

**The video is outdated:**
See at: https://youtu.be/Juht6XGLcu0?t=83

**Note in advance:**
I'm Windows "qualified". So apologies if I have seriously missed something here.

### How to reproduce

**Steps to reproduce the behavior:**
Compare instructions as per the manual.
Check the video against the latest release instructions.
Installing over a very old release compared to where it all sits today.

### Expected behavior

It all must be easy for Non-Linux people. :-) 

### Driver version

Not Applicable.

### Venus OS device type

Cerbo GX

### Venus OS version

Release Candidate: V3.00~39

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

18

### Connection type

Serial USB adapter to RS485

### Config file

```ini
Not Applicable.
```

### Relevant log output

```shell
Not Applicable.
```

### Any other information that may be helpful

Not Applicable.

mr-manuel, 2023-05-08:
Hi, the video is reffering to the latest official release. The documentation/wiki should be very well documented.

The problem you described is already known https://github.com/Louisvdw/dbus-serialbattery/issues/619

closed 2023-05-08 by mr-manuel

Jaco-TTT, 2023-05-08:
Mr-Manual.

The new script you sent, worked perfectly the 1st time. Took seconds.

Close it and thank you.

**Made these changes:**
[image]
[image]
[image]

Later I will also stop the charging reduction as it wastes time, Victron controls it perfectly as long as the cells are balanced.

mr-manuel, 2023-05-08:
Don't do that. Just copy this lines to your `config.ini` and insert your BMS.
https://github.com/Louisvdw/dbus-serialbattery/blob/88e5eea93331ae99dc28fcf8b5a40cb866e432e2/etc/dbus-serialbattery/config.default.ini#L179-L182
All other changes will get lost after the next update.

Jaco-TTT, 2023-05-08:
Done. And thank you.

Regards

https://github.com/Louisvdw/dbus-serialbattery/blob/88e5eea93331ae99dc28fcf8b5a40cb866e432e2/etc/dbus-serialbattery/config.default.ini#L179-L182