Louisvdw/dbus-serialbattery #625: Options to easily remove the drive and "disengage" the driver from the system.

Jaco-TTT, 2023-05-08:
### Is your feature request related to a problem? Please describe.

No

### Describe the solution you'd like

**Option to:**
1) Easily "disconnect" the BMS from controlling the system in case of an "emergency".
2) Option to remove the driver in its entirety from the Cerbo/Venus. Like a blank install.
3) Way to remove all older files.

### Describe alternatives you've considered

Tried to update the Cerbo to the latest release, files are still there.
Considered Factory Resetting the Cerbo, but was worried it would be for naught.
Considered buying a new Cerbo. :-)

### Additional context

We were off for 4 hours (load-shedding) from 10 am - 2 pm because the Cerbo "lost" the battery, and the inverter "off", with a second load-shedding scheduled 6 hours later at 8 am... we work from home.

There must be a "backdoor" for when the BMS connection is lost, for a non-issue, to disable the driver temporarily.

Especially when it was due to an install/configuration matter.

Jaco-TTT, 2023-05-08:
I saw this ... BLOCK_ON_DISCONNECT

That is maybe an option?

mr-manuel, 2023-05-08:
Which version are you using?

1. Just unplug the serial adapter. If you do not have set `BLOCK_ON_DISCONNECT = True` (default is `False)`, else you have to restart the driver/Venus OS once.
2. [Uninstall/remove the driver](https://louisvdw.github.io/dbus-serialbattery/general/install#uninstallremove-the-driver)
3. Which older files? On every update the folder is recreated.

Jaco-TTT, 2023-05-08:
Yes, I tried everything on that page more than once.

Regards
Jaco Reinecke

Jaco-TTT, 2023-05-08:
> * Just unplug the serial adapter. If you do not have set `BLOCK_ON_DISCONNECT = True` (default is `False)`, else you have to restart the driver/Venus OS once.
> 
I could not find BLOCK_ON_DISCONNECT anywhere. I saw it here on Github.
Restarted it +-15 times, with no effect.
What DID work, when LS was over, and I switched the inverter back on, THEN it all worked again with the driver, the one I tried every which way from Sunday to get it uninstalled.

> * 
[Uninstall/remove the driver](https://louisvdw.github.io/dbus-serialbattery/general/install#uninstallremove-the-driver)
> 
The removal commands had no effect at all.

> * Which older files? On every update the folder is recreated.
**Under:**
 /data/etc/dbus-serialbattery - I could only see files/folders dated 2022.
 /opt/victronenergy/dbus-serialbattery - I could see files/folders dated from last night.
 
**If I update the settings for my bank under:**
 /opt/victronenergy/dbus-serialbattery then the file I updated will show today's date here /data/etc/dbus-serialbattery 

I think I know what happened. 

Will try again tonight after LS.
Also got an idea, to stop the driver from starting, just add a character in like the config, so that it stops loading the driver.

mr-manuel, 2023-05-08:
You have to install the latest pre-release for `BLOCK_ON_DISCONNECT`. 

The files under `/data/etc/dbus-serialbattery` are the install files and copied over to `/opt/victronenergy/dbus-serialbattery` on every reboot or if you call `reinstall-local.sh`.

The files in `/data/etc/dbus-serialbattery` are Venus OS update persistent, the files in `/opt/victronenergy/dbus-serialbattery` not.

mr-manuel, 2023-05-08:
Since there are already `disable.sh` and `uninstall.sh` scripts I close this issue.

closed 2023-05-08 by mr-manuel

Jaco-TTT, 2023-05-08:
> Since there are already `disable.sh` and `uninstall.sh` scripts I close this issue.

Will test that.

Thank you.

Jaco-TTT, 2023-05-09:
> You have to install the latest pre-release for `BLOCK_ON_DISCONNECT`.
>
> The files under `/data/etc/dbus-serialbattery` are the install files and copied over to `/opt/victronenergy/dbus-serialbattery` [...]

Awesome will try that the moment I've got it all figured out.