Louisvdw/dbus-serialbattery #636: JBD BMS will not be recognized (reopen of #608)

sloesp, 2023-05-12:
### Describe the problem

This is a reopen of #608. Now with physical cells.

Hello,
i am using a Jaibaida AP21S002 21S 300A BMS with the RS485/USB Interface. If i use the JBDTools on Windows i get Data from the BMS. With Venus OS on RASP-Pi i see the LEDs on the Interface blinking for both directions (send/receive) but the are not BMS Information at the GUI.

Any Ideas?
Thanks so far.
Slotty

### Driver version

0.14.3

### Venus OS device type

Raspberry Pi

### Venus OS version

V2.92 same with 2.93

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

8

### Connection type

Serial USB adapter to RS485

### Config file

```ini
original utils.py
```

### Relevant log output

```shell
@400000006453a4fd23a7a5bc INFO:SerialBattery:Testing LltJbd
@400000006453a4fd2a21742c INFO:SerialBattery:Testing Ant
@400000006453a4fd3a038a24 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006453a4fd3a1600b4 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@400000006453a4fd3a1f99bc INFO:SerialBattery:Testing Daly
@400000006453a4fe0e6c9e8c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006453a4fe0e7ce684 INFO:SerialBattery:Testing Daly
@400000006453a4fe1e60bdb4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006453a4fe1e71154c INFO:SerialBattery:Testing Jkbms
@400000006453a4fe2e63b1bc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006453a4fe2e741124 INFO:SerialBattery:Testing Lifepower
@400000006453a4ff02c8074c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006453a4ff02d7c2a4 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@400000006453a4ff02e17aec INFO:SerialBattery:Testing Renogy
@400000006453a4ff12c2f4a4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006453a4ff12d3a22c INFO:SerialBattery:Testing Renogy
@400000006453a4ff22b22de4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006453a4ff22c2a8a4 INFO:SerialBattery:Testing Ecs
@400000006453a500085e150c ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyACM0
```

mr-manuel, 2023-05-12:
Hi, can you post the output of `ls -l /dev/tty*`?

mr-manuel, 2023-05-12:
Are you using a Raspberry Pi RS485 HAT? Can you please post pictures of your connection?

mr-manuel, 2023-05-12:
`/dev/ttyAMA0` normally is a Raspberry Pi RS485 HAT, therefore I asked.

Still waiting for pictures to help you.

Please try the beta version of the driver.

sloesp, 2023-05-13:
Here is the picture of my connections
[image]

sloesp, 2023-05-13:
With the nightly build installed i got the following log

@40000000645f693406911954 INFO:SerialBattery:Testing Jkbms
@40000000645f69341694a1cc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000645f6934169fc55c INFO:SerialBattery:Testing Lifepower
@40000000645f693426704b3c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000645f6934267b093c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@40000000645f69342681a88c INFO:SerialBattery:Testing LltJbd
@40000000645f69342cd3a3ac ERROR:SerialBattery:Unexpected err=UnicodeDecodeError('utf-8', b'JBD-AP21S002-L21-300A-B-U-R-C-\xdd', 30, 31, 'unexpected end of data'), type(err)=<class 'UnicodeDecodeError'>
@40000000645f69342cda7d94 INFO:SerialBattery:Testing Renogy
@40000000645f6935010df36c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000645f69350118dc64 INFO:SerialBattery:Testing Renogy

mr-manuel, 2023-05-13:
There were some issues with the JBD/LLT in the beta. Can you try this one?

https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases/tag/v1.0.20230515beta

Clericerr, 2023-05-15:
Hello,
I had the same problem with the JBD RS485 to USB converters. The problem is that these are recognized as ACM0 and are therefore interpreted by Venus as a partial GPS adapter. I was able to solve the problem by adapting the serial-starter.rules file.

1) read out the ID_SERIAL_SHORT from the converter with:
udevadm info --query=property --name=/dev/ttyACM0

2) open etc/udev/rules.d/serial-starter.rules
[image]

3) add your converter

ACTION=="add", ENV{ID_BUS}=="usb", ENV{ID_SERIAL_SHORT}=="Add your ID here",          ENV{VE_SERVICE}="sbattery"

[image]

sloesp, 2023-05-16:
@Clericerr
Thank you the hint, but it didnt help. Communication via RS485 Interface is working, but response causes an error due to unexpexted end of data.

Aquasada, 2023-05-25:
Something similar happens to me.
I've been using the driver with two bms jbd for about a year, after updating the driver to v0.14.3 all good, then update venus to 3.0 and good. Today I wanted to edit dbushelper.py to change the default name to the hardware version since I edited this field in the bms but when I restart it did not detect the bms. After uninstalling and installing now with version 1.0 of the driver and venus 2.94 I have managed to get one of the bms to appear, always the same although it usually takes time to recognize it, the windows application works fine

mr-manuel, 2023-05-25:
Could you please try with the latest beta released today? There were some changes in the JBD/LLT BMS driver.

https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v1.0.20230525beta

Aquasada, 2023-05-25:
I just installed it and it still doesn't detect, now neither of the two bms appears

Aquasada, 2023-05-25:
> ¿Podría intentarlo con la última versión beta lanzada hoy? Hubo algunos cambios en el controlador JBD/LLT BMS.
> 
> https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v1.0.20230525beta

I just installed it and it still doesn't detect, now neither of the two bms appears

Aquasada, 2023-05-25:
I have done several tests
I have installed version v0.9 without the bms appearing. Then install the latest final version v0.14.3 and here if the bms appear. I reboot a few times and it stops detecting them

mr-manuel, 2023-05-26:
@idstein maybe you can take a look?

idstein, 2023-05-26:
@Aquasada can you check out the latest dev version? There are a couple of new checks added to the communication with the device itself.
It should now better check the packet transmissions and validate the checksum.

idstein, 2023-05-26:
@Aquasada can you provide logs from the latest dev version?
Reading through @sloesp logs my guess is there is a missing boundary check for the name. It tries to decode the string with a wrong \xdd character at the end.

sloesp, 2023-05-26:
@idstein 
Here is my complett log with the latest build 

@40000000647093ab26161d4c INFO:SerialBattery:
@40000000647093ab2621ecbc INFO:SerialBattery:Starting dbus-serialbattery
@40000000647093ab26478d8c INFO:SerialBattery:dbus-serialbattery v1.0.20230525beta
@40000000647093ab2647a4fc INFO:SerialBattery:Testing Daly
@40000000647093ab2d98b0fc ERROR:SerialBattery:read_sentence 94: no sentence start received
@40000000647093ab2da23294 INFO:SerialBattery:request_data: bad reply no. 0
@40000000647093ab2daadd54 WARNING:SerialBattery:No data received in read_status_data()
@40000000647093ab34ed4b24 ERROR:SerialBattery:read_sentence 90: no sentence start received
@40000000647093ab34f6d48c INFO:SerialBattery:request_data: bad reply no. 0
@40000000647093ac00a06c5c ERROR:SerialBattery:read_sentence 90: no sentence start received
@40000000647093ac00aa111c INFO:SerialBattery:request_data: bad reply no. 0
@40000000647093ac07e95e24 ERROR:SerialBattery:read_sentence 57: no sentence start received
@40000000647093ac07f2b8ac INFO:SerialBattery:request_data: bad reply no. 0
@40000000647093ac07fa8c94 WARNING:SerialBattery:No data received in read_battery_code()
@40000000647093ac080fc62c INFO:SerialBattery:Testing Daly
@40000000647093ac0f6b1714 ERROR:SerialBattery:read_sentence 94: no sentence start received
@40000000647093ac0f74b01c INFO:SerialBattery:request_data: bad reply no. 0
@40000000647093ac0f7dd00c WARNING:SerialBattery:No data received in read_status_data()
@40000000647093ac16c05164 ERROR:SerialBattery:read_sentence 90: no sentence start received
@40000000647093ac16c9dacc INFO:SerialBattery:request_data: bad reply no. 0
@40000000647093ac1e177014 ERROR:SerialBattery:read_sentence 90: no sentence start received
@40000000647093ac1e178b6c INFO:SerialBattery:request_data: bad reply no. 0
@40000000647093ac2564edec ERROR:SerialBattery:read_sentence 57: no sentence start received
@40000000647093ac25650d2c INFO:SerialBattery:request_data: bad reply no. 0
@40000000647093ac256518e4 WARNING:SerialBattery:No data received in read_battery_code()
@40000000647093ac259be43c INFO:SerialBattery:Testing Ecs
@40000000647093ac290e008c INFO:SerialBattery:Testing HeltecModbus
@40000000647093ac291d4e84 INFO:SerialBattery:Testing on slave address 1
@40000000647093ac2e672c5c INFO:SerialBattery:found in try 1/10 for /dev/ttyACM0(1):     $^M5^LÑ
@40000000647093ac2e6747b4 G
@40000000647093ac2e674b9c y^M�^M5      ã
@40000000647093ac2e67536c G~ô~ô^H^A^@^@
@40000000647093ad0b412444 WARNING:SerialBattery:Error reading settings from BMS, retry (1/10): Checksum error in rtu mode: '\ta' instea
@40000000647093ad23ba08c4 WARNING:SerialBattery:Error reading settings from BMS, retry (2/10): Checksum error in rtu mode: '\ta' instea
@40000000647093ae0096a08c WARNING:SerialBattery:Error reading settings from BMS, retry (3/10): Checksum error in rtu mode: '\ta' instea
@40000000647093ae19127adc WARNING:SerialBattery:Error reading settings from BMS, retry (4/10): Checksum error in rtu mode: '\ta' instea
@40000000647093ae318eca5c WARNING:SerialBattery:Error reading settings from BMS, retry (5/10): Checksum error in rtu mode: '\ta' instea
@40000000647093af0e6be6f4 WARNING:SerialBattery:Error reading settings from BMS, retry (6/10): Checksum error in rtu mode: '\ta' instea
@40000000647093af26de012c WARNING:SerialBattery:Error reading settings from BMS, retry (7/10): Checksum error in rtu mode: '\ta' instea
@40000000647093b003b49db4 WARNING:SerialBattery:Error reading settings from BMS, retry (8/10): Checksum error in rtu mode: '\ta' instea
@40000000647093b01c28bb8c WARNING:SerialBattery:Error reading settings from BMS, retry (9/10): Checksum error in rtu mode: '\ta' instea
@40000000647093b01c329314 INFO:SerialBattery:None
@40000000647093b01c7d0d7c Traceback (most recent call last):
@40000000647093b01c7d2104   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 156, in <module>
@40000000647093b01c7d348c     main()
@40000000647093b01c7d3c5c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 125, in main
@40000000647093b01c7d4fe4     battery = get_battery(port)
@40000000647093b01c7d57b4   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 83, in get_battery
@40000000647093b01c7d6b3c     if battery.test_connection():
@40000000647093b01c7e2e8c   File "/opt/victronenergy/dbus-serialbattery/bms/heltecmodbus.py", line 101, in test_connection
@40000000647093b01c7e45fc     and self.read_status_data()
@40000000647093b01c7e51b4   File "/opt/victronenergy/dbus-serialbattery/bms/heltecmodbus.py", line 229, in read_status_data
@40000000647093b01c7e653c     logger.info("Heltec-" + self.hwTypeName)
@40000000647093b01c7e70f4 AttributeError: 'HeltecModbus' object has no attribute 'hwTypeName'

mr-manuel, 2023-05-26:
@sloesp please use the full command, when you get logs, see [How to troubleshoot](https://louisvdw.github.io/dbus-serialbattery/troubleshoot/). Else we cannot see the correct time, which also helps us to understand the problem better.

When posting code/logs please put tripple backticks \`\`\` in the line before and after the code. I edited your post.

idstein, 2023-05-26:
> @idstein Here is my complett log with the latest build
> 

The culprit is a broken HeltecModbus BMS implementation. Can you try to add `BMS_TYPE = LltJbd` to your `config.ini` and capture the logs a gain?

sloesp, 2023-05-26:
@mr-manuel 
Thank you for the hint to create useful logfiles but in this case there are several controlcharacters in the log so the output didnt contain the hole log.

sloesp, 2023-05-26:
Here is the log with configured BMS_TYPE in config.ini

```
2023-05-26 12:08:06.223343500 #4
2023-05-26 12:08:06.223344500 #5
2023-05-26 12:08:06.223346500 #6
2023-05-26 12:08:06.223347500 #7
2023-05-26 12:08:07.408779500 INFO:SerialBattery:
2023-05-26 12:08:07.409209500 INFO:SerialBattery:Starting dbus-serialbattery
2023-05-26 12:08:07.410104500 INFO:SerialBattery:dbus-serialbattery v1.0.20230525beta
2023-05-26 12:08:07.410542500 INFO:SerialBattery:Testing LltJbd
2023-05-26 12:08:07.882757500 ERROR:SerialBattery:Unexpected err=UnicodeDecodeError('utf-8', b'JBD-AP21S002-L21-300A-B-U-R-C-\xdd', 30, 31, 'unexpected end of data'), type(err)=<class 'UnicodeDecodeError'>
2023-05-26 12:08:08.384041500 INFO:SerialBattery:Testing LltJbd
2023-05-26 12:08:08.853816500 ERROR:SerialBattery:Unexpected err=UnicodeDecodeError('utf-8', b'JBD-AP21S002-L21-300A-B-U-R-C-\xdd', 30, 31, 'unexpected end of data'), type(err)=<class 'UnicodeDecodeError'>
2023-05-26 12:08:09.355065500 INFO:SerialBattery:Testing LltJbd
2023-05-26 12:08:09.829465500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-05-26 12:08:10.096669500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-05-26 12:08:10.370917500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-05-26 12:08:10.873097500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyACM0
2023-05-26 12:08:10.873253500 #0
2023-05-26 12:08:10.873255500 #1
2023-05-26 12:08:10.873256500 #2
2023-05-26 12:08:10.873258500 #3

```

idstein, 2023-05-26:
@mr-manuel there are 2 fixes for review which helps to resolve this issue here.

mr-manuel, 2023-05-26:
I merged them. @sloesp can you try to install the nightly from the `dev` branch? Run the `install.sh` and select `3`and then `2`.

sloesp, 2023-05-26:
Problem is fixed. BMS is recognized also without being configured in the config.ini

closed 2023-05-26 by sloesp