Louisvdw/dbus-serialbattery #638: utils.py overwritten after reboot

nighteagle1974, 2023-05-12:
### Describe the bug

Hello,

have change the settings in the utils.py with nano - i use nano often on my linux-machines so i know how to handle it.
First time i change the settings in utils.py and then i wonder why it is on default again - so i have a look and see the values back to default. I have then change again - check double the file with nano - make something an check again - all fine. Then i make reboot - voila - utils.py is back to default. Then i have copied utils.py to utils.org - then change settings and rename - the same after reboot all settings back to default. I have read the wiki but it looks like a simple nano change file thing - but not?

### How to reproduce

Steps to reproduce the behavior:
1. Open utils.py with nano over SSH
2. Change comething
3. Reboot
4. Check utils.py again

### Expected behavior

Persistant changes

### Driver version

0.14.3

### Venus OS device type

Cerbo GX

### Venus OS version

v2.92

### BMS type

Daly Smart BMS

### Cell count

8S

### Connection type

Serial USB adapter to TTL

### Config file

```ini
Not neccesary
```

### Relevant log output

```shell
Not neccesary
```

nighteagle1974, 2023-05-12:
Have found the problem - if you have SD-Card inside the CERBO GX with the File _**venus-data.tar.gz**_ - my first thing to install - but wiki tells you to install only over SSH on Cerbo GX. So i forgot the SD-Card inside Cerbo GX.
Looks like it is unpacking and overwrite files afte reboot with this file on SD-Card.

For the MOD can you write this into WIKI to prevent other people run into this.

nighteagle1974, 2023-05-12:
Is USB-Install same as SD-CARD install - and is the wiki correct that i can install on CERBO GX only over SSH? Or is somthing old info? Thought if it can not install over SD-CARD then it is okay to ignore the file first... but not. And a SD-CARD is so little and hidden.. in the Cerbo-Slot - a USB-Stick is big and i see it the whole time :-)

mr-manuel, 2023-05-12:
[Why is my utils.py always reset to default values?](https://louisvdw.github.io/dbus-serialbattery/faq/#why-is-my-utilspy-always-reset-to-default-values)

> Is USB-Install same as SD-CARD install - and is the wiki correct that i can install on CERBO GX only over SSH? Or is somthing old info? Thought if it can not install over SD-CARD then it is okay to [...]

It seems that it works now. I will change that.

closed 2023-05-12 by mr-manuel