Louisvdw/dbus-serialbattery #640: BMS SOC vs. Multiplus SOC

kr0815, 2023-05-12:
### Describe the problem

In the VRM-Portal, i see the SOC provided by d-bus-serialbattery and the one provided by the internal meassurement

If i change the BMS SOC, the Mutiplus SOC will also be changed?

Is this behaviour wanted?

### Driver version

1.0.0 installed with nightly built

### Venus OS device type

Raspberry Pi

### Venus OS version

3.00 36

### BMS type

Daly Smart BMS

### Cell count

16

### Connection type

Serial USB adapter to TTL

### Config file

```ini
*
```

### Relevant log output

```shell
*
```

mr-manuel, 2023-05-28:
> If i change the BMS SOC, the Mutiplus SOC will also be changed?

Is this a statement or a question?

From what I know, the SoC of the battery monitor selected should be displayed. Where exactly in the VRM portal do you mean? Please post a screenshot.

kr0815, 2023-05-28:
In VRM, i see the Values of the Daly BMS - read out by serialbattery, and, theoretically, the value calculated by the shunt in the Multiplus

But it looks like serialbattery is also overwriting the internal Multiplus value?

[image]

mr-manuel, 2023-05-28:
In the remote console/GUI go to Settings -> System setup -> Battery measurements and enable the devices you want to see in VRM.

kr0815, 2023-05-28:
I have them both enabled 

Problem is, that internal Value is always the same as the one from serialbattery

I think there is a bug that serialbattery overwrites also the internal calculated SOC?

mr-manuel, 2023-05-28:
No, the driver is not overwriting nothing. Probably it‘s some calculation in [`dbus-systemcalc-py`](https://github.com/victronenergy/dbus-systemcalc-py).

Please open a question in the Victron Community.

kr0815, 2023-05-28:
Thanks, i´ll try that
Really strange behaviour, as vor Voltage and Current, i get different Valus, but SOC is always the same

mr-manuel, 2023-05-28:
Please link your post here, so other users with the same problem can follow that

closed 2023-05-30 by mr-manuel