Louisvdw/dbus-serialbattery #644: tar: conf/serial-starter.d: Cannot open: File exists

buddhafragt, 2023-05-14:
### Describe the bug

I got this error message after "bash /tmp/install.sh":

tar: conf/serial-starter.d: Cannot open: File exists
tar: Exiting with failure status due to previous errors

How to solve?

### How to reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

installation not possible

### Driver version

latest

### Venus OS device type

Raspberry Pi

### Venus OS version

v2.92

### BMS type

JKBMS / Heltec BMS

### Cell count

4

### Connection type

Serial USB adapter to RS485

### Config file

```ini
installation not possible
```

### Relevant log output

```shell
installation not possible
```

### Any other information that may be helpful

installation not possible

mr-manuel, 2023-05-14:
Which version are you trying to install?

mr-manuel, 2023-05-22:
In theory this error could only occur, if you installed a beta version and then you tried to install over a version below `v1.0.0`.

Any update on this or can I close it?