Louisvdw/dbus-serialbattery #647: Linear Limitation goes above MAX_CELL_VOLTAGE

Joeknx, 2023-05-16:
### Describe the bug

If I turn on Linear Limitation and CVCM I see MAX_CELL_VOLTAGE 55.36 but in config.ini it is set to 55.2 (3.450)

### How to reproduce

[image]

### Expected behavior

Keep MAX_CELL_VOLTAGE from the config.ini

### Driver version

v1.0.20230516beta

### Venus OS device type

Raspberry Pi

### Venus OS version

3.0.0-18

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 90.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; Example with 1 BMS: Jkbms_Ble C8:47:8C:00:00:00
; Example with 3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
BLUETOOTH_BMS =

; --------- BMS disconnect behaviour ---------
; Description: Block charge and discharge when the communication to the BMS is lost. If you are removing the
;              BMS on purpose, then you have to restart the driver/system to reset the block.
; False: Charge and discharge is not blocked on BMS communication loss
; True: Charge and discharge is blocked on BMS communication loss, it's unblocked when connection is established
;       again or the driver/system is restarted
BLOCK_ON_DISCONNECT = False

; --------- Charge mode ---------
; Choose the mode for voltage / current limitations (True / False)
; False is a step mode: This is the default with limitations on hard boundary steps
[... 173 lines cut ...]

; -- LltJbd settings
; SoC low levels
; NOTE: SOC_LOW_WARNING is also used to calculate the Time-To-Go even if you are not using a LltJbd BMS
SOC_LOW_WARNING = 20
SOC_LOW_ALARM   = 10

; -- Daly settings
; Battery capacity (amps) if the BMS does not support reading it
BATTERY_CAPACITY = 50
; Invert Battery Current. Default non-inverted. Set to -1 to invert
INVERT_CURRENT_MEASUREMENT = 1

; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT = 15
```

### Relevant log output

```shell
...
```

mr-manuel, 2023-05-16:
Absorption dynamic means, that one cell has hit the max cell voltage and therefore an automated dynamic voltage is calculated. Can you post also a screenshot of the cell voltages, if it‘s still in dynamic?

Joeknx, 2023-05-16:
Ok, if it's a normal behavior it's fine. I updated from an older version, tried the settings and was a bit confused.
The cells were all at max cell voltage (Diff 0,003) for at least 5 hours. 
I can post a screenshot tomorrow but it's probably not necessary.

mr-manuel, 2023-05-16:
It could be a bug, so it’s better check ;-)

Joeknx, 2023-05-17:
When the CVL rises (7:40), the cells had a difference of approx 0,030V. Highes Cell 3,471, lowest 3,441.
When it raised again (7:51) the cells had a difference of approx 0,020V. Highes Cell 3,468, lowest 3,449.

Looks ok to me. 
[image]
[image]
[image]

closed 2023-05-17 by mr-manuel