Louisvdw/dbus-serialbattery #648: EG4 LifePower bugs

najaB76, 2023-05-17:
### Describe the bug

On trying out the beta driver I get an error saying that MAX_BATTERY_CURRENT is not defined.

```2023-05-17 16:23:49.433061500     self.max_battery_current = utils.MAX_BATTERY_CURRENT
2023-05-17 16:23:49.433065500 AttributeError: module 'utils' has no attribute 'MAX_BATTERY_CURRENT'```

This is in dbus-serialbattery/bms/lifepower.py - should be MAX_BATTERY_CHARGE_CURRENT

Then I get the errors:

```
2023-05-17 16:27:38.818836500 Traceback (most recent call last):
2023-05-17 16:27:38.818845500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 371, in publish_battery
2023-05-17 16:27:38.818850500     self.battery.manage_charge_current()
2023-05-17 16:27:38.818853500   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 392, in manage_charge_current
2023-05-17 16:27:38.818859500     ccl = round(min(charge_limits), 3)  # gets changed after finished testing
2023-05-17 16:27:38.818863500 TypeError: '<' not supported between instances of 'NoneType' and 'float'
```
and 
```
2023-05-17 16:30:57.145394500 Traceback (most recent call last):
2023-05-17 16:30:57.145401500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 371, in publish_battery
2023-05-17 16:30:57.145406500     self.battery.manage_charge_current()
2023-05-17 16:30:57.145409500   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 412, in manage_charge_current
2023-05-17 16:30:57.145414500     self.charge_limitation = charge_limits[min(charge_limits)]
2023-05-17 16:30:57.145418500 TypeError: '<' not supported between instances of 'NoneType' and 'float'
```

Which I have 'fixed' by changing the values to 60 for both.  I presume that it's not pulling the correct value from the config file, but it's not immediately obvious where it should be getting them from (more of a PHP/JavaScript guy than Python),

### How to reproduce

Install version 	1.0.20230508beta on top of version 0.14.3 using install.sh

### Expected behavior

Driver should load without throwing errors and respect the charge current limits as per the config file.

### Driver version

1.0.20230508beta

### Venus OS device type

Raspberry Pi

### Venus OS version

v2.93

### BMS type

Life/Tian Power

### Cell count

8

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 50.0
[... 346 lines cut ...]
2023-05-17 16:32:23.697189500 INFO:SerialBattery:> Connection voltage: 27.63V | Current: 0.0A | SoC: 100.0%
2023-05-17 16:32:23.697701500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 8
2023-05-17 16:32:23.698227500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-05-17 16:32:23.698758500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 60.0A
2023-05-17 16:32:23.699380500 INFO:SerialBattery:> CVCM:     True
2023-05-17 16:32:23.699915500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2023-05-17 16:32:23.700443500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2023-05-17 16:32:23.700965500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-05-17 16:32:23.701493500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2023-05-17 16:32:23.745824500 INFO:SerialBattery:DeviceInstance = 2
2023-05-17 16:32:23.746674500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2023-05-17 16:32:23.778065500 INFO:SerialBattery:Hardware Version:B
2023-05-17 16:32:23.827806500 INFO:SerialBattery:Firmware Version:3QT-YS00-16SV100A-V3.5
2023-05-17 16:32:23.846835500 INFO:SerialBattery:publish config values = 1
2023-05-17 16:32:54.162159500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-05-17 16:32:54.162877500 ERROR:SerialBattery:>>> ERROR: Incorrect Data
2023-05-17 16:35:44.351298500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-05-17 16:35:44.352157500 ERROR:SerialBattery:>>> ERROR: Incorrect Data
2023-05-17 16:38:34.546522500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-05-17 16:38:34.547409500 ERROR:SerialBattery:>>> ERROR: Incorrect Data
2023-05-17 16:41:24.749853500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-05-17 16:41:24.750739500 ERROR:SerialBattery:>>> ERROR: Incorrect Data
2023-05-17 16:44:14.950743500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-05-17 16:44:14.951650500 ERROR:SerialBattery:>>> ERROR: Incorrect Data
```

mr-manuel, 2023-05-17:
Thanks for the report, I will look at it

mr-manuel, 2023-05-18:
All errors was caused by this wrong two variables.

https://github.com/Louisvdw/dbus-serialbattery/blob/88e5eea93331ae99dc28fcf8b5a40cb866e432e2/etc/dbus-serialbattery/bms/lifepower.py#L41

Could you please try this release and confirn that the issue was solved or not? https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases/tag/v1.0.20230519beta

najaB76, 2023-05-18:
Installed that version and no issues so far.

Thanks for the quick fix.

closed 2023-05-19 by mr-manuel